JSBSim-Team/jsbsim #135: Python bindings

Phillip-M-Feldman, 2018-12-11:
I have a question about the following:

"A Python module of JSBSim can also be built by CMake. For that, you need Cython installed on your platform. CMake will automatically detect Cython and build the Python module."

I don't fully understand this, and am hoping for a bit more explanation.  Also, I noticed that the Python bindings have not been updated in many years.  Are they compatible with Python 3.0?

seanmcleod, 2018-12-11:
The bindings are compatible with Python 3, they were updated a while ago (6 months?) to support both Python 3 and Python 2. The CI builds for Linux build for both Python 2&3 while the CI build for Windows only builds for Python 3.

If you don't have Cython installed then you'll need to install it, e.g. if you take a look at appveyor.yml which is used for the Windows CI builds you'll notice a comment stating that in the AppVeyor environment Cython isn't installed by default so we explicitly install it via `pip install` before running CMake etc.

`pip install Cython numpy pandas scipy`

The references to `numpy pandas scipy` are there since our Python based tests use them, they aren't required for simply building the JSBSim Python module.

Phillip-M-Feldman, 2018-12-11:
I do have Cython installed, and will give this a try.  Thanks!

I got concerned because the page at
http://jsbsim-python.sourceforge.net/index.html shows a last update of
2007, which creates the impression of an inactive project.

Phillip-M-Feldman, 2018-12-11:
FYI, the page at https://sourceforge.net/projects/jsbsim-python/files/ also
shows a 2007 date.

seanmcleod, 2018-12-11:
JSBSim development moved to GitHub a while ago, some but not all sourceforge pages were updated to point people to GitHub.

Phillip-M-Feldman, 2018-12-11:
I've noticed that the JSBSim Python demo depends on pandac.  I haven't been
able to find source code or descriptive material for that package.  Any
pointers will be appreciated.

seanmcleod, 2018-12-12:
Which demo are you referring to? This one - [JSBSim Python Bindings - Demo](http://jsbsim-python.sourceforge.net/demos.html) refers to Panda3D but has an explicit link to Panda3D.

bcoconni, 2018-12-15:
> I've noticed that the JSBSim Python demo depends on pandac.

May be you mean [pandas](https://pandas.pydata.org) ? It is a Python data analysis library which basically serves to manipulate huge amount of data as JSBSim can produce while running a simulation.

Phillip-M-Feldman, 2018-12-17:
JSBSim-Demo.py has a comment to the effect that it depends on Panda3D
1.3.2.  For some reason, the code imports from `pandac` rather than
`panda3d`.

Phillip

bcoconni, 2018-12-22:
> JSBSim-Demo.py has a comment to the effect that it depends on Panda3D 1.3.2.

You should rather ask the author of `JSBSim-Demo.py` we are not providing support to the interface of JSBSim with Panda3D.

bcoconni, 2019-03-17:
@Phillip-M-Feldman 
I think this issue originated from a confusion between an external inactive project and our own Python bindings. As I said earlier we are not providing support for this external support.

However we are providing support for our own Python bindings. Have you managed to use the Python module generated by JSBSim ?

bcoconni, 2019-03-24:
Connection lost. Issue closed.

closed 2019-03-24 by bcoconni