JSBSim-Team/jsbsim #137: Has anyone gotten the Boeing314 or any other than Rascal, JSBSim Aircraft to run in github jsbsim installed and ardupilot mavproxy?

josheeg, 2018-12-20:
Has anyone gotten the Boeing314 or any other than Rascal, JSBSim Aircraft to run in github jsbsim installed and ardupilot mavproxy? It won't takeoff after mode auto arm throttle
The instructions for the new JSBSim github version is out it shows sourceforge but I got it to run in both sourceforge and git hub using git. http://ardupilot.org/dev/docs/setting-up-sitl-on-linux.html
[image]

WillemEerland, 2018-12-21:
Nice project! Quite sure I'm not interested in boarding, but someone else might :) it probably is a lot cheaper if you can replace the pilot with a raspberry PI.....

Anyways, about your question: First off, you can get more information about what JSBSim is doing by adding the flag "--debug" when you run sim_vehicle.py. This will provide an additional terminal with information about what JSBSim is doing.

And while I don't have specific experience with flying a Boeing, I'm familiar with running models other than the Rascal. To understand how ArduPilot is actually loading JSBSim, have a look at the file "libraries/SITL/SIM_JSBSim.cpp" (ardupilot) and you can see how the startup script is generated. Take for example the following part, where you can see how one engine is set to running. This script may need to be altered depending on your aircraft configuration.

```
"    <event name=\"start engine\">\n"
"      <condition> simulation/sim-time-sec le 0.01 </condition>\n"
"      <set name=\"propulsion/engine[0]/set-running\" value=\"1\"/>\n"
"      <notify/>\n"
"    </event>\n"
```

I would try and replace it with the following, for which I got inspiration by looking at the "/scripts/B737_Runway.xml" (jsbsim) script. 

```
"    <event name=\"start engine\">\n"
"      <condition> simulation/sim-time-sec le 0.01 </condition>\n"
"      <set name=\"propulsion/starter_cmd\" value=\"1\"/>\n"
"      <notify/>\n"
"    </event>\n"
```

We'll be able to give you more specific advice once you provide the feedback from JSBSim.

josheeg, 2018-12-22:
Thank you,  Yea I might, the rasbery pi does have some impressive computer vison and high level libraries like flytos.
[flytos](http://ardupilot.org/dev/docs/flytos.html)
 
I ran the --debug in my command "sim_vehicle.py -f jsbsim:Rascal --console --map --debug -L Ballarat"
Also "sim_vehicle.py -f jsbsim:Boeing314 --console --map --debug -L Ballarat"
It compiled SITL with the debug option also it shown more verbose output. But did not open another window. 

The Rascal flew the mission the Boeing314 did not move no issues with mode auto arm throttle if I gave both of them a minute between the commands. 

So you have run other models than the Rascal are they JSBSim releases? Or are they models put into aeromatic++ from OpenVSP? I just picked a few that I recognized the name.  ... Concorde, x24b, and Camel. They didn't work. Are there easy electric motor ones that work? Quadplane?

Those startup scripts are interesting.

WillemEerland, 2018-12-22:
"The screen the JSBSim 's output is on scrolls fast with the seconds the simulator runs so it is hard to read or copy or debug." - This is exactly the screen I'm talking about. You can pause the simulation via the ArduPilot terminal to review ( `param set SIM_SPEEDUP 0` ) .

The standard control of JSBSim via ArduPilot is rather limited (JSBSim can do much more) - it only converts the PWM signals to input commands and some settings controlling the wind velocity ( I think these need an update ). See this code snippet:

```
             "set fcs/aileron-cmd-norm %f\n"
             "set fcs/elevator-cmd-norm %f\n"
             "set fcs/rudder-cmd-norm %f\n"
             "set fcs/throttle-cmd-norm %f\n"
             "set atmosphere/psiw-rad %f\n"
             "set atmosphere/wind-mag-fps %f\n"
             "set atmosphere/turbulence/milspec/windspeed_at_20ft_AGL-fps %f\n"
             "set atmosphere/turbulence/milspec/severity %f\n"
             "iterate 1\n",
```

so really it's the startup script that needs to initialise the aircraft at the right locations, and do a ground trim ( which, if you see it on the map, it does successfully ) and prepare it for takeoff, i.e., start the engine! ( here lies the problem... ).

Try updating the startup script. I'll have a look later on to see if I can't update the script with something that works on the other aircraft, but this update will be done on the ArduPilot side of the code.

josheeg, 2018-12-22:
I modified SIM_JSBSim.cpp with the part above and get this error I saved it as SIM_JSBSim_backup.cpp
to make it easy to bring it back. 
 
engine[0]/set-running\" value=\"1\"/>\n"

---- JSBSim Execution beginningOpened JSBSim control socket
 ... --------------------------------------------

FATAL ERROR: JSBSim terminated with an exception.
The message was: No property, "propulsion/starter_cmd" is defined.
Fatal: Failed to send on control socket: Connection reset by peer

josheeg, 2018-12-22:
I wonder if their is just a template that would be needed since I got these files from jsbsim release.
Like they were parsed or expected to be moved to different places.
Boeing314.xml  Engines  INSTALL  reset.xml is all that is in Boeing314 
Engines$ ls
HamiltonStd6243A-3.xml  WrightGR-2600.xml

Now you see in Rascal..

Dialogs                   Rascal110-splash.rgb  README.Rascal       Systems
Engines                   Rascal-keyboard.xml   reset_CMAC.xml
Models                    Rascal-submodels.xml  reset_template.xml
Rascal110-JSBSim-set.xml  Rascal.xml            reset.xml

18x8.xml  Zenoah_G-26A.xml

This is kinda useful because I am more interested in electric motors when I make a drone with OpenVSP. 

I wonder if the JSBSim files were made with a older aeromatic possibly web aeromatic vs aeromatic++???

I may try copying the Rascal and remaining it Boeing314 and move the specific Boeing314 files over then not modify any code and try it?

josheeg, 2018-12-22:
I tried entering in the command line 
param set SIM_SPEEDUP 0
pasted it and hit enter and the 5 second time cound the simulator is running still makes it hard to see what is going on can it be piped to a text file or log?

josheeg, 2018-12-22:
I ran the tests and installed the python module
also the python dependencies mentioned here

https://github.com/JSBSim-Team/jsbsim
~/jsbsim/build$ 
ctest -j4

Test project /home/josheeg/jsbsim/build
      Start  1: ResetOutputFiles
      Start  2: TestICOverride
      Start  3: RunCheckCases
      Start  4: TestModelLoading
 1/52 Test  #2: TestICOverride ...................   Passed    1.25 sec
      Start  5: TestActuator
 2/52 Test  #1: ResetOutputFiles .................   Passed    1.45 sec
      Start  6: TestGustReset
 3/52 Test  #6: TestGustReset ....................   Passed    2.73 sec
      Start  7: TestPointMassInertia
 4/52 Test  #5: TestActuator .....................   Passed    3.79 sec
      Start  8: CheckMomentsUpdate
 5/52 Test  #8: CheckMomentsUpdate ...............   Passed    0.66 sec
      Start  9: TestFuelTanksInertia
 6/52 Test  #7: TestPointMassInertia .............   Passed    4.41 sec
      Start 10: TestInputSocket
 7/52 Test  #9: TestFuelTanksInertia .............   Passed    4.82 sec
      Start 11: TestInitialConditions
 8/52 Test #10: TestInputSocket ..................   Passed    3.40 sec
      Start 12: CheckScripts
 9/52 Test #11: TestInitialConditions ............   Passed   13.34 sec
      Start 13: CheckAircrafts
10/52 Test  #4: TestModelLoading .................   Passed   27.11 sec
      Start 14: CheckOutputRate
11/52 Test #13: CheckAircrafts ...................   Passed    3.70 sec
      Start 15: TestAccelerometer
12/52 Test #14: CheckOutputRate ..................   Passed    1.01 sec
      Start 16: CheckDebugLvl
13/52 Test #16: CheckDebugLvl ....................   Passed   12.42 sec
      Start 17: TestCosineGust
14/52 Test #17: TestCosineGust ...................   Passed    1.85 sec
      Start 18: TestScriptOutput
15/52 Test #18: TestScriptOutput .................   Passed    1.59 sec
      Start 19: CheckSimTimeReset
16/52 Test #19: CheckSimTimeReset ................   Passed    0.90 sec
      Start 20: TestHoldDown
17/52 Test #15: TestAccelerometer ................   Passed   21.90 sec
      Start 21: CheckTrim
18/52 Test #21: CheckTrim ........................   Passed    1.74 sec
      Start 22: TestChannelRate
19/52 Test #22: TestChannelRate ..................   Passed    1.30 sec
      Start 23: TestWaypoint
20/52 Test #20: TestHoldDown .....................   Passed    8.22 sec
      Start 24: TestSuspend
21/52 Test #23: TestWaypoint .....................   Passed    0.69 sec
      Start 25: TestLGearSteer
22/52 Test #24: TestSuspend ......................   Passed    1.19 sec
      Start 26: TestAeroFuncOutput
23/52 Test #25: TestLGearSteer ...................   Passed    1.03 sec
      Start 27: TestKinematic
24/52 Test #27: TestKinematic ....................   Passed    1.20 sec
      Start 28: TestTurboProp
25/52 Test #12: CheckScripts .....................   Passed   45.41 sec
      Start 29: TestEngineIndexedProps
26/52 Test #28: TestTurboProp ....................   Passed    4.71 sec
      Start 30: TestExternalReactions
27/52 Test #26: TestAeroFuncOutput ...............   Passed    8.41 sec
      Start 31: TestTurbine
28/52 Test  #3: RunCheckCases ....................   Passed   65.38 sec
      Start 32: TestAeroFuncFrame
29/52 Test #31: TestTurbine ......................   Passed    7.26 sec
      Start 33: TestTemplateFunctions
30/52 Test #33: TestTemplateFunctions ............   Passed    0.93 sec
      Start 34: TestDensityAltitude
31/52 Test #34: TestDensityAltitude ..............   Passed    0.68 sec
      Start 35: TestPressureAltitude
32/52 Test #35: TestPressureAltitude .............   Passed    0.68 sec
      Start 36: fpectl
33/52 Test #36: fpectl ...........................   Passed    0.57 sec
      Start 37: TestStdAtmosphere
34/52 Test #37: TestStdAtmosphere ................   Passed    1.18 sec
      Start 38: TestMiscellaneous
35/52 Test #38: TestMiscellaneous ................   Passed    0.61 sec
      Start 39: TestGndReactions
36/52 Test #39: TestGndReactions .................   Passed    0.64 sec
      Start 40: TestIntegrators
37/52 Test #40: TestIntegrators ..................   Passed    0.84 sec
      Start 41: TestSwitch
38/52 Test #41: TestSwitch .......................   Passed    0.68 sec
      Start 42: TestGain
39/52 Test #30: TestExternalReactions ............   Passed   16.99 sec
      Start 43: TestDeadBand
40/52 Test #42: TestGain .........................   Passed    0.64 sec
      Start 44: TestFilter
41/52 Test #43: TestDeadBand .....................   Passed    0.68 sec
      Start 45: TestFunctions
42/52 Test #44: TestFilter .......................   Passed    0.74 sec
      Start 46: TestDistributor
43/52 Test #46: TestDistributor ..................   Passed    0.75 sec
      Start 47: FGColumnVector3Test1
44/52 Test #47: FGColumnVector3Test1 .............   Passed    0.03 sec
      Start 48: StringUtilitiesTest1
45/52 Test #48: StringUtilitiesTest1 .............   Passed    0.10 sec
      Start 49: FGJSBBaseTest1
46/52 Test #49: FGJSBBaseTest1 ...................   Passed    0.10 sec
      Start 50: FGMatrix33Test1
47/52 Test #45: TestFunctions ....................   Passed    1.25 sec
      Start 51: FGQuaternionTest1
48/52 Test #50: FGMatrix33Test1 ..................   Passed    0.05 sec
      Start 52: FGLocationTest1
49/52 Test #51: FGQuaternionTest1 ................   Passed    0.20 sec
50/52 Test #52: FGLocationTest1 ..................   Passed    0.21 sec
51/52 Test #29: TestEngineIndexedProps ...........   Passed   44.33 sec
52/52 Test #32: TestAeroFuncFrame ................   Passed  212.62 sec

100% tests passed, 0 tests failed out of 52

Total Test time (real) = 278.05 sec

seanmcleod, 2018-12-22:
Take the startup script and any initialization file generated by Ardupilot SITL and use them to run the model just using JSBSim on the command line stand-alone. You'll then be able to easily see the output that JSBSim produces with regards to any errors with the startup/initialization scripts.

seanmcleod, 2018-12-22:
Also don't forget that the Boeing 314 has 2 engines that you want to start.

One option is to use `<running> -1 </running>` in an initial conditions file to get all irrespective of how many engines there are running.

There was a fix checked in about 6 months ago to fix a bug with this option, see - [Handle -1 for all engines running during intialization](https://github.com/JSBSimTeam/jsbsim/commit/d5fefe82aa39454dbb27dd17cbb462c2cb442df0)

seanmcleod, 2018-12-22:
Taking a quick glance at ArduPilot's [SIM_JSBSim.cpp](https://github.com/ArduPilot/ardupilot/blob/6cbbbcd51dbc227c815c9178f140882ae0c5640e/libraries/SITL/SIM_JSBSim.cpp) I'd suggest maybe looking into removing the `start engine` event from the script generated and replacing it with `<running> -1 </running>`  in the reset/initialization file that Ardupilot generates.

In addition to handling the starting of multiple engines I'm pretty sure this will also handle the differences in how you start a piston engine versus a turbofan engine etc.

josheeg, 2018-12-22:
> Take the startup script and any initialization file generated by Ardupilot SITL and use them to run the model just using JSBSim on the command line stand-alone. You'll then be able to easily see the output that JSBSim produces with regards to any errors with the startup/initialization scripts.

I got JSBSim to run its help file it didn't seem to like what I put in. 

I do not know what file you  mean by the startup script. 
I also do not know what file you mean as a initialization file.

JSBSim --aircraft=/home/josheeg/ardupilot/Tools/autotest/aircraft/Rascal/Rascal.xml --script=/home/josheeg/jsbsim/scripts/b171.xml

seanmcleod, 2018-12-22:
@josheeg take a look at [SIM_JSBSim.cpp](https://github.com/ArduPilot/ardupilot/blob/6cbbbcd51dbc227c815c9178f140882ae0c5640e/libraries/SITL/SIM_JSBSim.cpp).

On line 85 it creates a JSBSim script file and then populates it through lines 90 to 123. This is the file I referred to as the 'startup script'.

Now also note that on line 100 it references an initialization file which it names as `reset`. This is the file I referred to as 'any initialization file'. This file is generated from line 151 to 161.

josheeg, 2018-12-23:
I see 
reset.xml
a initialization file
Is the only one I can find.
I have  SIM_JSBSim.cpp
I see the Eng. section 90-123
But I do not see the file in Rascal folder also

seanmcleod, 2018-12-23:
@josheeg if you take a look at SIM_JSBSim.cpp you'll see that `reset.xml` is created in the aircraft directory.

```c++
asprintf(&jsbsim_reset, "%s/aircraft/%s/reset.xml", autotest_dir, jsbsim_model);
```
However the startup script file isn't created in the aircraft directory.

```c++
asprintf(&jsbsim_script, "%s/jsbsim_start_%u.xml", autotest_dir, instance);
```

josheeg, 2018-12-23:
ok now I see the file thanks. So these two files your saying 

jsbsim_start_0.xml

Now I will try to put 

<running> -1 </running>

where all of this is.... I find it written over and I had to enter arm throttle with the Rascal so I think it was written over. 

<event name="start engine">
      <condition> simulation/sim-time-sec le 0.01 </condition>
      <set name="propulsion/engine[0]/set-running" value="1"/>
      <notify/>
    </event>

What do you suggest? Running them in command line with jsbsim? what good would that do?

josheeg@josheeg-Vostro-220-Series:~/ardupilot/Tools/autotest$ ls
aircraft                         fg_plane_view.bat
ap1.txt                          fg_plane_view.sh
apmrover2.py                     fg_quad_view.bat
arducopter.py                    fg_quad_view.sh
ArduPlane-Missions               flaps.txt
arduplane.py                     __init__.py
ArduRover-Missions               jsb_sim
ardusub.py                       jsbsim_fgout_0.xml
autotest.py                      jsbsim_start_0backup.xml
balancebot1.txt                  jsbsim_start_0.xml
balancebot.py                    junit.xml
bisect-helper.py                 locations.txt
build-with-disabled-features.py  param_metadata
CMAC-circuit.txt                 plane-gripper-mission.txt
CMAC-grid.txt                    pysim
common.py                        quadplane.py
copter_AVC2013_mission.txt       README
copter_glitch_mission.txt        rover1.txt
copter-gripper-mission.txt       rover-camera-mission.txt
copter_loiter_to_alt.txt         rover-fence-ac-avoid.txt
copter_mission.txt               rover-gripper-mission.txt
copter_nav_delay_takeoff.txt     run_in_terminal_window.sh
copter_nav_delay.txt             sim_vehicle.py
copter_payload_place.txt         sub-gripper-mission.txt
copter_spline_mission.txt        sub_mission.txt
copter_terrain_mission.txt       web
default_params                   web-firmware
examples.py                      win_sitl
fakepos.py                       XPlane

WillemEerland, 2019-03-05:
Have a look at 

[https://github.com/WillemEerland/ardupilot/tree/jsbsim](https://github.com/WillemEerland/ardupilot/tree/jsbsim)

and use ardupilot (control via MavProxy) to fly a 737.

`ardupilot/ArduPlane/sim_vehicle.py --frame=jsbsim:737 --location=KSFO --debug`

notes:
- make sure the latest JSBSim is in the path (e.g., export PATH=<location>:$PATH)
- I used [QGroundControl](http://qgroundcontrol.com/) and used manual control to add thrust and see the aircraft move

Questions?

josheeg, 2019-03-07:
>    - make sure the latest JSBSim is in the path (e.g., export PATH=:$PATH)

I will try that I appreciate the tip.

closed 2020-12-11 by bcoconni