JSBSim-Team/jsbsim #138: Feature Request: dynamically drag of propeller

HHS81, 2018-12-20:
Actually it isn`t possible to simulate icing of propellers in jsbsim. For that it it necessary to have a possibility to input/ modify the drag coefficient created by the ice on blades, so we get the typical effects of propeller icing. Expected effect:
non-constant speed prop: loss of rpm
constant-speed-prop: blade-pitch reduce, thrust reduce, engine power and prop rpm doesn`t change.

bcoconni, 2018-12-22:
That's an interesting request but it will request quite some work I think. Please be patient 😃.

hbeni, 2024-03-15:
> That's an interesting request but it will request quite some work I think. Please be patient 😃.

Hi,
any news on this?