JSBSim-Team/jsbsim #140: FlightGear: useless warnings

mike402, 2018-12-22:
Hi, I had posted this on FlightGear mailing list before I realised these warnings are actually in JSBSim source. I hope my claims are valid and I'm not misusing the systems too badly?

https://sourceforge.net/p/flightgear/mailman/message/36499706/
> When I start FG with some aircraft which has a lot of JSBSim code, for
> example my Tu-144, I see a lot of warnings that are just noise that
> drowns out really important things like the new warnings about missing
> <function> args, they just get no chance of being seen unless you look
> for them on purpose. When I get crash because of a missing property I
> also have to grep through the logs for possible offenders, because
> it's not always mentioned in the last few lines.
> 
> 1. "Missing" properties
> 
> FGPropertyManager::GetNode() No node found for fcs/canard/pos-norm-input
> 
> presumably caused by properties that are a filter with name="" that
> another filter created before the given one tries to read, but fails
> the first time because it's not there yet
> 
> This warning normally serves a good purpose everywhere else, but
> JSBSim will just exit if it finds anything that is "really" missing.
> 
> It might be that replacing name="" with <output> would fix it, but I
> can not imagine the amount of editing in many files of many aircraft,
> because I might not be the only one to have been using name="" for too
> long before finding out it causes this.
> 
> 2.
> 
> In file /usr/home/mike/src/git/ac/Tu-144/Tu-144-jsbsim/control.xml: line 180
>       Property fcs/canard/pos-norm is already defined.
> 
> These are of course about the properties that I have defined in the
> beginning of each JSBSim system as <property> that are replaced by
> filters with name="" on the same property name, in this case
> "fcs/canard/pos-norm".
> 
> I think this practice of defining <property> I came up with is a good
> one (and hope it catches on if it's not terrible misuse) and should
> not be discouraged by such warnings because:
> - this, interspersed with some comments, is a good way to clearly
>   document the input interface of the system
> - it allows you to remove for debugging some system, without editing
>   a single line in all the systems depending on it (otherwise you have
>   to either add every property used by the other system as <property>
>   or comment out everything that uses it -- which could be hundreds of
>   lines in dozens of places)
> - another good use is to define initial values for properties there
>   (such as panel switches) or dummy value for isolated testing,
>   because it's self-contained. The only other option is maintaining a
>   copy in -set.xml, which would be a huge pain to keep in sync.

Cheers,
Mike

bcoconni, 2018-12-23:
Indeed these warnings are not linked to FlightGear. But this might need some work to make this warning messages context sensitive.

mike402, 2018-12-23:
Maybe it would make sense to only fire those warnings after the very first iteration, when the property tree has settled? But then the warnings about one thing replaching the other won't work...

NikolaiVChr, 2018-12-23:
**FGPropertyManager::GetNode() No node found for ...**

3 ways to fix this:

1. If that node is declared in JSB then be sure that component that declares it is before the component that uses it.
If above cannot be done due to cross referenced properties:
2. Declare the property in JSB with `<property>` (not good solution, cause will give warnings when relocating (reinit) the aircraft.)
3. Declare the property in the -set file. The set file is always run before any JSB system.
4. It can also indicate a typo somewhere, another reason to keep this warning.

**Property ... is already defined**

If you use JSB from FG then firstly do not declare any properties with `<property>` in JSB. That will give this warning when reinit. Do it in -set file.

If you get this warning when starting up FG, then you have done something bad. Like writing to it from multiple components will give you this warning (I think, from the top of my head).

Anyway, heeding these warning makes the JSB /-set file more tidy when you fix them. In my opinion an aircraft should give zero JSB warnings when starting up or reinit.

mike402, 2018-12-23:
On Sat, Dec 22, 2018 at 07:13:47PM -0800, Nikolai V. Chr. wrote:
> **FGPropertyManager::GetNode() No node found for ...**
> 
> 3 ways to fix this:
> 
> 1. If that node is declared in JSB then be sure that component that declares it is before the component that uses it.
> If above cannot be done due to cross referenced properties:

How about when you have many many systems that interdepend on each other?

> 2. Declare the property in JSB with `<property>` (not good solution, cause will give warnings when relocating (reinit) the aircraft.)

Probably would also cause the "already defined" warning I mentioned.

> 3. Declare the property in the -set file. The set file is always run before any JSB system.

See below.

> 4. It can also indicate a typo somewhere, another reason to keep this warning.

I already said that if somethingg is *really* missing, JSBSim will exit. So this warning is just noise.

> **Property ... is already defined**
> 
> If you use JSB from FG then firstly do not declare any properties with `<property>` in JSB. That will give this warning when reinit. Do it in -set file.

See below.

> If you get this warning when starting up FG, then you have done something bad. Like writing to it from multiple components will give you this warning (I think, from the top of my head).

"Something bad" is when you create 2 filters with the same path. Cause is there really anything wrong with a filter writing to or replacing some <property>? Why not count <property> as just "input var declaration" that makes sure the thing exists?

> Anyway, heeding these warning makes the JSB /-set file more tidy when you fix them. In my opinion an aircraft should give zero JSB warnings when starting up or reinit.

Tidy? Hello! Can you imagine how painful and error-prone would syncing -set.xml against some huge JSBSim model would be? This is a very, very ugly crutch! Also what about standalone JSBSim then, or JSBSim in some other hypothetical sim -- would you have to sync it with yet another file? Of course you do need -set.xml for *some* things like some initial positions of switches -- but why recreate the whole JSBSim tree there just because?

https://upload.wikimedia.org/wikipedia/commons/e/e0/Forgotten_edits_in_copypaste_programming.gif

bcoconni, 2018-12-27:
@NikolaiVChr 
The purpose is not to suppress these warnings messages altogether but rather to make sure that they are issued only when relevant.

mike402, 2018-12-28:
> @mike402
> The last commit to FlightGear might alos possibly reduce the number of spurious warnings. JSBSim now makes an extended use of its class `FGPropertyValue` for sensors which allows late binding and should hopefully save some warnings about missing properties.
> 
> Let me know if you notice a difference with regard to that topic.

Thanks, the 1st kind of errors (No node found) are gone completely! Now I only have double definition ones.

mike402, 2019-08-01:
What about the remaining warning in `FGPropertyReader.cpp` around line `Property ... is already defined`?

There is really no way to go without `<property>` declarations in the system files, because as soon as you have to remove some dependency system to test your system's inputs, you would have to add all those missing properties somewhere, instead of simply commenting out a system.

mike402, 2019-08-01:
By the way, it might be a separate issue but `<table>` of `<fcs_function><function>` requires its inputs to be declared as `<property>`, even if such filter exists and even if it exists in the same system and makes FG exit if it is not declared.

bcoconni, 2019-08-04:
> What about the remaining warning in `FGPropertyReader.cpp` around line `Property ... is already defined`?
>
> There is really no way to go without `<property>` declarations in the system files,

I think you are referring to this piece of code ?
https://github.com/JSBSim-Team/jsbsim/blob/de6bbb1fa0785142dafe5eb3825967d28bd641d7/src/input_output/FGPropertyReader.cpp#L87-L110
If so, then I think you missed the point of the test which is to make sure that properties are not defined twice unless explicitly allowed by the flag `override`.

And the privilege of overriding properties definition is only given to scripts `FGScripts` to allow local modifications of the settings without altering the aircraft definition files.

bcoconni, 2019-08-04:
> By the way, it might be a separate issue but `<table>` of `<fcs_function>` `<function>` requires its inputs to be declared as `<property>`, even if such filter exists and even if it exists in the same system and makes FG exit if it is not declared.

Could you please provide an example of the problem you are describing ? What is the current way of declaring `<table>` in `<fcs_function>` that you would expect should be done differently ?

bcoconni, 2019-12-01:
@mike402 
I think the 2nd kind of warning message has also been considerably reduced since the components are no longer tied to the property specified in `name=""` ? For the record this feature has been pushed in FlightGear since Aug 10, 2019.

mike402, 2019-12-02:
I still see a million "already defined" warnings unless I comment out this message:

https://github.com/JSBSim-Team/jsbsim/blob/master/src/input_output/FGPropertyReader.cpp#L105

bcoconni, 2019-12-04:
> I still see a million "already defined" warnings unless I comment out this message

Well, unfortunately we are facing a problem here.

When the message `<property> already defined` is issued, JSBSim ignores the statement at stake and iterates to the next property definition. Our general policy is that when a statement is purposely ignored then a warning message is issued so that the user knows that something (possibly unexpected) is going on. From that point of view, removing the warning message is not an option. And even more so since we were bitten quite hard by the `<and>` and `<or>` statements that were silently misinterpreted by JSBSim and led to a debate with FlightGears contributors about its resolution (see issue #181)

Another option would be to no longer ignore the property definition statement and do "something" with it. But whatever the "something" is, it would be another breach to the backward compatibility of JSBSim, and FlightGear users and devs are becoming quite reluctant to any further such breaches since we have thrown a couple of them across their legs in the recent past.

The only way out I can imagine would be to classify the messages in different categories (information, warning, error, ...) and allow the user to filter whichever messages he/she is interested in. I think that is what the simgear logging infrastructure `SG_LOG` does. Its integration in JSBSim is probably quite easy but the task of parsing each message issued by JSBSim and classify them among information, warning, error is definitely not something that can be done overnight.

Unless someone has another idea ?

mike402, 2019-12-04:
Imho, it only makes sense to issue this warning only if the previous
definition of that property was another filter -- not simply a
<property> -- because in that case no logic gets hurt. So most of
those messages I get are just noise, and I can't see if I really have
filters conflicting for some name.

And no, don't give me that "remove all those <property>" again because
firstly, it's some (loose but ok) kind of input specification and
secondly saves a lot of time when you want to:
- test the system by manually setting its inputs;
- comment out some <system> without editing 100500 things that depend
  on its properties.
Does absolutely no one else find what I'm talking about useful?

On Wed, Dec 04, 2019 at 11:27:35AM -0800, Bertrand Coconnier wrote:
> Well, unfortunately we are facing a problem here.
>
> When the message `<property> already defined` is issued, JSBSim ignores the statement at stake and iterates to the next property definition. Our [...]

bcoconni, 2019-12-04:
@mike402 
> And no, don't give me that "remove all those <property>" again

Did I ever say/write that ? Could you point at the message where I asked you to "remove all those <property>" ?

This kind of untruthful statement does definitely not serve your case. So I am kindly encouraging you to reconsider your position if you want this discussion to proceed any further.

Thank you.

mike402, 2019-12-04:
I did not say you did, Nikolai did. I only mean to say I am frustrated that this remains the only way to avoid these warnings.

> If you use JSB from FG then firstly do not declare any properties with `<property>` in JSB. That will give this warning when reinit. Do it in -set file.

bcoconni, 2019-12-05:
> I did not say you said that, Nikolai said it. I only mean to say I am frustrated that this remains the only way to avoid these warnings.
> > If you use JSB from FG then firstly do not declare any properties with <property> in JSB. That will give this warning when reinit. Do it in -set file.

I am pretty sure @NikolaiVChr was merely suggesting you one possible way (among others) to address these warning messages and was not requesting you to remove the properties.

Anyway, regarding the technical side of the subject you have to understand that:
* Properties listed at the start of systems or channels are meant to be **property definitions** i.e. ment to create a non existing property and set its initial value to the value specified by the attribute `value=""`
```xml
<system>
  <property value="-2.0"> my_property </property> <!-- initialize "my_property" to -2.0 -->
```
* If the attribute `value=""` is not specified, JSBSim initializes the property to 0.0
* If the property already exists then the statement `<property value="-2.0"> my_property </property>` is ignored and JSBSim issues a message.
---
Now say that the warning message is removed as per your request. Sooner or later, a user will open an issue saying "*In the code below, the statement `<property value="42"> prop2 </property>`seems to have no effect. The value 42 is ignored, and the property `prop2` is equal to 2.0.*"
```xml
<system>
  <property value="1.0"> prop1 </property>
  <channel name="channel">
    <pure_gain name="prop2">
      <input> prop1 </input>
      <gain> 2.0 </gain>
    </pure_gain>
  </channel>
</system>
<system>
    <property value="42"> prop2 </property>
```
What should I tell to that user ? *JSBSim ignores that statement because the property `prop2` has already been defined in the other system. We used to have a warning message that would have helped you spot the problem but we removed it because some users found it annoying. How unfortunate. If you had read the C++ code you would have known.*

The conclusion of this example is that this warning message is neither *useless* nor *noise* as you qualified it. The mere fact that it annoys you is not an indication that it will be of no use to anyone. A number of alternate options have been suggested along this discussion, you might even suggest new ones but the warning message will not be removed.

mike402, 2019-12-05:
Sorry, I forgot the `value=""` would get ignored, message is indeed useful in that case.

Without `value=""`, though, as user I would expect that a property gets "declared" to make sure things don't quit if nothing else is created there (e.g. something has been commented out). That it gets filled with 0.0 is very handy because it keeps things from exploding -- but is just so by jsbsim's design, not explicitly intended.

How about suppressing the warning for `<property>`, but not for `<property value="something">`? Is there an easy enough way to tell `<property>` apart from `<property value="0.0">` so that it wouldn't break the latter?

mike402, 2020-08-31:
Why not simply use the already-existing value of <propert> to initialize the element (for example integrator or actuator) with that value? That would also help FlightGear implement full state saving later.

bcoconni, 2020-08-31:
> Why not simply use the already-existing value of to initialize the element (for example integrator or actuator) with that value? That would also help FlightGear implement full state saving later.

Because:
* This is not backward compatible, which is a definitive no-go
* Bugs due to uninitialized variables are difficult to track down
* FlightGear is not JSBSim only usage.

Regarding the initialization of actuators, this has already been discussed in issue #295 (which applies to integrators too).

bcoconni, 2020-09-06:
@mike402 
After giving more thoughts to your request, I think I jumped hastily to a false conclusion. Apologies for that.

You are suggesting to use the existing value of a property to initialize an element but this is already what JSBSim does: if the property exists, it is used unmodified and if doesn't exist it is created (and set to 0.0 unless told otherwise). So I don't understand why you are requesting that as a modification ? Could you be more specific about your request ? Is it not redundant with the topic in issue #295 which has been resolved ?

mike402, 2020-09-06:
From what I understand it is not, because in #295 the request was to make <input> go through to output?

What I am suggesting is, using the existing property value to set the initial output value of the element that is replacing that property. E.g. if an `<actuator name="blabla">` is replacing `<property value="123.45">blablabla</property>`, its initial position would be `123.45`.

closed 2023-03-24 by bcoconni