JSBSim-Team/jsbsim #141: Unuseable fuel - make it do something?

legoboyvdlp, 2018-12-24:
Hi,
It would seem <unusable-gal_us> in the -set.xml of a FlightGear aircraft does nothing, as the engine keeps ticking over even below this level - is this something JSBSIM can handle automatically or should we hack our fuel system?

legoboyvdlp, 2018-12-29:
It should be a matter of cutting off fuel if fuel is below that level.

The thing is, can JSB access a property from FlightGear?

bcoconni, 2018-12-29:
> The thing is, can JSB access a property from FlightGear?

Sure, you must provide the absolute path of the property, something like `/consumables/fuel/tank[1]/unusable-gal_us`.

legoboyvdlp, 2018-12-30:
My apologies - I meant, if it were to be included as a built in feature of jsbsim would that be possible? I understand it would probably not work with JSBSIM standalone?

bcoconni, 2018-12-30:
> if it were to be included as a built in feature of jsbsim would that be possible? I understand it would probably not work with JSBSIM standalone?

It's perfectly feasible to add that feature to JSBSim standalone and interface it with the property `unusable-gal_us` of FlightGear.

bcoconni, 2018-12-31:
> It's perfectly feasible to add that feature to JSBSim standalone and interface it with the property `unusable-gal_us` of FlightGear.

I pushed the commits b8f4a4c and c8c5c6e to JSBSim to account for the unusable volume of fuel in the tanks. I also pushed these changes to FlightGear and updated the interface between FlightGear and JSBSim so that the values specified via the properties `<unusable-gal_us>` in the `-set.xml` are propagated to the fuel management of JSBSim.

Could you please test this new feature and let me know if that works as expected ?

legoboyvdlp, 2018-12-31:
Hi,
That's great - I'll build and test later on this week.

seanmcleod, 2019-01-03:
@legoboyvdlp taking a look at @bcoconni's commit in `FGPropulsion.cpp` he does appear to set the engine to starved if none of the tanks have fuel greater than the unusable amounts.

```c++
bool Starved = true; // Initially set Starved to true. Set to false in code below.
…
        case FGTank::ttFUEL:
          if ((Tank->GetContents() > Tank->GetUnusable()) && Tank->GetSelected() && (TankPriority == CurrentFuelTankPriority)) {
            TanksWithFuel++;
            Starved = false;
            FeedListFuel.push_back(TankId);
          } 
          break;
…
bool FuelStarved = Starved;
…
engine->SetStarved(FuelStarved || (hasOxTanks && OxiStarved));
```
I haven't double-checked the engine code yet to see whether it handles the `SetStarved()` call correctly. Which engine type are you testing with currently?

seanmcleod, 2019-01-03:
So in `FGPiston::doFuelFlow()`:

```c++
  if(Starved) // There is no fuel, so zero out the flows we've calculated so far
  {
    equivalence_ratio = 0.0;
    FuelFlowRate = 0.0;
    m_dot_fuel = 0.0;
  }
  FuelFlow_pph = FuelFlowRate  * 3600;
```

So `FuelFlow_pph == 0` when the engine is starved.

In `FGPiston::doEnginePower()`:

```c++
IndicatedHorsePower = (FuelFlow_pph / ISFC )* ME * power - StaticFriction_HP;
```

So the indicated HP should be equal to `-StaticFriction_HP`, and by default `StaticFriction_HP = 1.5;`

And in `FGPiston::Calculate()`:

```c++
  doEnginePower();
  if (IndicatedHorsePower < 0.1250) Running = false;
```

So when the engine is starved `Running` should be set to `false`.

I'm assuming you're checking the `Running` property when you say the engine 'keeps running'?

legoboyvdlp, 2019-01-03:
No actually, I just don't see the engine sound changing, or the rpm indicator moving.

legoboyvdlp, 2019-01-03:
I start with 2.0 gal.

After a few minutes full throttle it is down to 1.5 gallons, after which fuel consumption stops.
However,

fdm/jsbsim/propulsion/engine[0]/engine-rpm remains at 2038; fuel-flow-rate-gph at just over 11 gallons.

Fuel does not reduce below 1.5 gallons after ten minutes.

/engines/engine[0]/running is true.

seanmcleod, 2019-01-04:
Okay so I ran a quick test. I'm using just plain JSBSim, latest commit to the master branch etc., i.e. I'm not using FlightGear.

So I made the following fuel tank changes to `c172p.xml`:

```xml
        <tank type="FUEL">    <!-- Tank number 0 -->
            <location unit="IN">
                <x> 56 </x>
                <y> -112 </y>
                <z> 59.4 </z>
            </location>
            <capacity unit="LBS"> 185 </capacity>
            <contents unit="LBS"> 100 </contents>
            <unusable-volume unit="GAL"> 15.14 </unusable-volume>
        </tank>
        <tank type="FUEL">    <!-- Tank number 1 -->
            <location unit="IN">
                <x> 56 </x>
                <y> 112 </y>
                <z> 59.4 </z>
            </location>
            <capacity unit="LBS"> 185 </capacity>
            <contents unit="LBS"> 100 </contents>
            <unusable-volume unit="GAL"> 15.14 </unusable-volume>
        </tank>
```

Basically to have just enough usable fuel for a couple of seconds. Then I have a script which sets the aircraft up trimmed at 4,000ft and reports a number of properties once a second.

```bash
Time Notify (Event 2) executed at time: 15.008273
    fcs/throttle-cmd-norm = 0.664446
    propulsion/engine/power-hp = 67.285959
    propulsion/engine/engine-rpm = 2087.760568
    propulsion/engine/fuel-flow-rate-gph = 5.104828
    propulsion/engine/set-running = 1.000000

Time Notify (Event 2) executed at time: 16.008269
    fcs/throttle-cmd-norm = 0.664446
    propulsion/engine/power-hp = -3.507435
    propulsion/engine/engine-rpm = 1769.555108
    propulsion/engine/fuel-flow-rate-gph = 0.000000
    propulsion/engine/set-running = 0.000000

Time Notify (Event 2) executed at time: 17.008265
    fcs/throttle-cmd-norm = 0.664446
    propulsion/engine/power-hp = -2.492419
    propulsion/engine/engine-rpm = 1398.134029
    propulsion/engine/fuel-flow-rate-gph = 0.000000
    propulsion/engine/set-running = 0.000000
```

So there is enough fuel for just over 15s worth, the usable fuel runs out sometime between 15s and 16s.

Once the usable fuel runs out you can see that `fuel-flow-rate-gph` drops to 0 and `set-running` toggles from 1 to 0, the engine HP drops and the engine rpm starts dropping.

After 50s, which was the time period for my script, the engine rpm has dropped to 664rpm.

```bash
Time Notify (Event 2) executed at time: 50.008133
    fcs/throttle-cmd-norm = 0.664446
    propulsion/engine/power-hp = 2.586441
    propulsion/engine/engine-rpm = 664.673636
    propulsion/engine/fuel-flow-rate-gph = 0.000000
    propulsion/engine/set-running = 0.000000
```

So everything appears to be working as expected in my setup.

legoboyvdlp, 2019-01-04:
Wait - I defined it in the -set.xml file, not in the FDM file. So this makes a difference?

legoboyvdlp, 2019-01-07:
I tested again, this time defining it in c172p.xml:
` <tank type="FUEL">
            <location unit="IN">
                <x> 56 </x>
                <y> -112 </y>
                <z> 59.4 </z>
            </location>
            <capacity unit="LBS">129.43</capacity>
            <contents unit="LBS">100.0 </contents>
			<unusable-volume unit="GAL">1.5</unusable-volume>
            <type>AVGAS</type>
        </tank>

        
        <tank type="FUEL">
            <location unit="IN">
                <x> 56 </x>
                <y> 112 </y>
                <z> 59.4 </z>
            </location>
            <capacity unit="LBS">129.43</capacity>
            <contents unit="LBS">100.0 </contents>
			<unusable-volume unit="GAL">1.5</unusable-volume>
            <type>AVGAS</type>
        </tank>

        
        <tank type="FUEL">
            <location unit="IN">
                <x> 56 </x>
                <y> 0 </y>
                <z> 59.4 </z>
            </location>
            <capacity unit="LBS">0.1</capacity>
            <contents unit="LBS">0.0</contents>
            <priority>1</priority>
            <type>AVGAS</type>
        </tank>

        
        <tank type="FUEL">
            <location unit="IN">
                <x> 56 </x>
                <y> 0 </y>
                <z> 59.4 </z>
            </location>
            <capacity unit="LBS">0.1</capacity>
            <contents unit="LBS">0.0</contents>
            <priority>1</priority>
            <type>AVGAS</type>
        </tank>`

Once the fuel reaches 9.0 lbs (1.5 gal), consumption stops - however,
 propulsion/engine/set-running is equal to ''

Power, thrust, and rpm all remain constant - thus I think the engine is still on?

seanmcleod, 2019-01-07:
Try commenting out tanks 2 and 3 and see what happens. 

My guess is there is some bug or interaction with those tanks which prevents `Starved` being set even after consumption from tank 1 stops once it gets down to 1.5 gal. How are you using tanks 2 and 3? Are they being fed fuel from somewhere?

The code snippet I posted earlier (see below) performs - `(Tank->GetContents() > Tank->GetUnusable()` which should return `false` for tanks 2 and 3 based on your contents of 0 above assuming they aren't being fed fuel. Which means they shouldn't be counted as tanks with fuel to prevent `Starved` being set to `true`.

```c++
        case FGTank::ttFUEL:
          if ((Tank->GetContents() > Tank->GetUnusable()) && Tank->GetSelected() && (TankPriority == CurrentFuelTankPriority)) {
            TanksWithFuel++;
            Starved = false;
            FeedListFuel.push_back(TankId);
          } 
          break;

```

legoboyvdlp, 2019-01-07:
2 and 3 are feed tanks, fed from 0 and 1.

bcoconni, 2019-01-12:
@legoboyvdlp 
The problem is coming from the c172p fuel system that drains fuel from the tanks to the float chamber

https://github.com/c172p-team/c172p/blob/ccc5462c628108dc000e8c5053141ea4c159dbeb/Systems/fuel.xml#L45
```xml
    <channel name="To Tank2">
        <!-- from Tank 0 (to Collector Tank 2) -->
        <switch name="fuel/from-tank0-to-tank2">
            <default value="0"/>
            <test logic="AND" value="0.1">
                propulsion/tank[0]/priority EQ 1 
                propulsion/tank[0]/contents-lbs GT 0
                /consumables/fuel/tank[2]/level-lbs LT 0.05
                propulsion/tank[2]/priority EQ 1
                accelerations/Nz GE 0
            </test>
        </switch>

        <!-- from Tank 1 (to Collector Tank 2) -->
        <switch name="fuel/from-tank1-to-tank2">
            <default value="0"/>
            <test logic="AND" value="0.1">
                propulsion/tank[1]/priority EQ 1 
                propulsion/tank[1]/contents-lbs GT 0
                /consumables/fuel/tank[2]/level-lbs LT 0.05
                propulsion/tank[2]/priority EQ 1
                accelerations/Nz GE 0
            </test>
        </switch>
    </channel>
```
Fuel is drained as long as `propulsion/tank[*]/contents-lbs` is greater than `0.0`. The contents should be compare to the volume of unusable fuel. I have submitted the PR c172p-team/c172p#1254 to fix the problem.

bcoconni, 2019-04-14:
@legoboyvdlp 
Now that the PR c172p-team/c172p#1254 has been merged in the c172p project, I guess this issue can be closed ?

legoboyvdlp, 2019-04-14:
Yes, indeed - thanks.

closed 2019-04-14 by legoboyvdlp