JSBSim-Team/jsbsim #142: How the wind effects on the aircraft dynamic are calculated ?

alexandrequ, 2018-12-29:
How the wind effects on the aircraft dynamic are calculated in JSBSim ? Is it just a global mean effect on the aircraft or the wind is integrate all along the wings ?

bcoconni, 2018-12-29:
> How the wind effects on the aircraft dynamic are calculated in JSBSim ?

This really depends on the implementation of the aircraft model. Most aircraft assume that the wind velocity is uniform across the wing span but as @seanmcleod pointed out some other aircraft models implement a much more complicated interaction.

> Is there only has a single input vector (i.e set of x,y,z wind velocities) for wind speed

The JSBSim atmosphere model can provide 2 vectors: 

- One for the wind direction (in north, east, down directions)
- One for the wind turbulence which is basically a rotation velocity in rad/sec

If you need a more complicated model, you will have to code it yourself and supply the wind velocity to JSBSim via the properties `/fdm/jsbsim/atmosphere/wing-down-fps`,  `/fdm/jsbsim/atmosphere/wing-east-fps` and `/fdm/jsbsim/atmosphere/wing-north-fps`.

If you want to use C++ code then this is discussed in issue #132.

alexandrequ, 2018-12-30:
Thank you for your fast answers. 

In my case, I will integrate the wind velocities along the wings from my data and then define mean values to use in `SetWindNED()`

But I also would like to add the roll-moment inducted by the wind. I found on the website I should add this to my "aircraftName.xml"
`     <function name="aero/coefficient/Cly">`
             ` <description>Roll_moment_due_to_wake_vortex</description>`
                `  <product>`
                    `   <property>aero/qbar-psf</property>`
                     `   <property>metrics/Sw-sqft</property>`
                   `   <property>metrics/bw-ft</property>`
                    `   <property>aero/stability_coefficients/Cl-y</property>`
                 `  </product>`
           `</function>`

But where should I add that part ?
`JSBSim::FGPropertyManager* node =  FDMExec->GetPropertyManager()->GetNode("aero/stability_coefficients/Cl-y")`
`node->setDoubleValue(paramVal);`

bcoconni, 2019-01-13:
@alexandrequ 
It is not clear whether your code exports wind velocities or resultant forces and moments on the aircraft. What I understand is that your wind code can be interfaced with C++.

## Option 1 - Importing wind velocities to JSBSim
If your external code computes wind velocities then you need to modify JSBSim code in `src/models/atmosphere/FGWinds.cpp` to call your external program.

The code that you will need to modify will be around line 147:
https://github.com/JSBSim-Team/jsbsim/blob/70428dc24db82d4697cc1cbd013c5aad99539112/src/models/atmosphere/FGWinds.cpp#L139-L155

There you can add your wind velocities to the total wind velocities `vTotalWindNED`. Assuming your code can be called by the C++ function `myWindFunction` this could be something like
```c++
  FGColumnVector3 myWindNED;
  
  myWindFunction(...); // assuming it returns wind velocities (wn, we, wd) in the NED frame

  myWindNED(eNorth) = wn;
  myWindNED(eEast) = we;
  myWindNED(eDown) = wd;

  vTotalWindNED = myWindNED + vWindNED + vGustNED + vCosineGust + vTurbulenceNED;
```
**Important note:** the wind velocity is expressed in the local frame (or NED for North, East, Down). More details about the NED frame definition [can be found in our docs](https://jsbsim-team.github.io/jsbsim-reference-manual/mypages/user-manual-frames-of-reference/#local-vertical-local-level-frame-or-local-ned-frame).

## Option 2 - Importing forces and moments to JSBSim
If your code generates the resultant forces and moments applied to the aircraft then you are better calling your code from `src/models/FGAircraft.cpp` which collects all the forces applied on the aircraft.

You will have to insert your code after the line 107:
https://github.com/JSBSim-Team/jsbsim/blob/70428dc24db82d4697cc1cbd013c5aad99539112/src/models/FGAircraft.cpp#L90-L112

The modifications you will need to make are very similar to what I described for the wind velocities. I assume that you have a C++ function `myForceFunction` that returns a force `fx, fy, fz` and another C++ function `myMomentFunction` that returns a moment `mx, my, mz` at the aircraft center of gravity
```c++
  FGColumnVector3 myForce;

  myForceFunction(...);

  myForce(eX) = fx;
  myForce(eY) = fy;
  myForce(eZ) = fz;

  vForces += myForce;

  FGColumnVector3 myMoment;

  myMomentFunction(...);

  myMoment(eX) = mx;
  myMoment(eY) = my;
  myMoment(eZ) = mz;

  vMoments += myMoment;
```
In that case the force and moment must be expressed in the aircraft body frame which description [can be found in our docs](https://jsbsim-team.github.io/jsbsim-reference-manual/mypages/user-manual-frames-of-reference/#body-frame).

alexandrequ, 2019-01-13:
Thanks for your answer @bcoconni !

bcoconni, 2019-02-09:
@alexandrequ 
Has your problem been fixed ?

alexandrequ, 2019-02-10:
Yes thank you

alexandrequ, 2019-02-12:
I have still one question, how can I access to the lift coefficient CL value to use it in the new function that I've created ? Thanks for the help

bcoconni, 2019-02-15:
> how can I access to the lift coefficient CL value to use it in the new function that I've created ?

The value of the CL coefficient can be computed from `src/models/FGAerodynamics.cpp`
https://github.com/JSBSim-Team/jsbsim/blob/8ade730a1c86c01f0cc3156ffceedcafbfa554c7/src/models/FGAerodynamics.cpp#L141-L150

JSBSim does only compute CL^2 because it is generally used to compute the lift induced drag CDi. However it is quite easy to modify that code to compute CL and export it to your code.

alexandrequ, 2019-02-20:
Thank you @bcoconni

alexandrequ, 2019-03-01:
@bcoconni I still have some difficulties to find in which file is the resulting forces due to vTotalWindNED are calculated ?

seanmcleod, 2019-03-01:
@alexandrequ take a look at line 148 in FGAuxiliary.cpp.

https://github.com/JSBSim-Team/jsbsim/blob/06f4ce37619d73c619c6296d1f602b10088bd95d/src/models/FGAuxiliary.cpp#L146-L149

alexandrequ, 2019-03-03:
Thank you for the information @seanmcleod ! 
Do you know where is vAeroUVW used to calculate the forces ?

seanmcleod, 2019-03-03:
It isn't used directly to calculate forces. It changes the values of alpha and beta and so any forces and moments defined in the aircraft's aerodynamics section that use alpha and beta will return different forces and moments due to the wind's effect on alpha and beta.

alexandrequ, 2019-03-04:
Thanks @seanmcleod ! But in which file is calculated the force with alpha ?

seanmcleod, 2019-03-04:
Are you familiar with how forces and moments are defined for JSBSim aircraft? If not take a look at the JSBSim manual and also browse through some of the aircraft configuration files.

So for example here is a snippet from the 737 model included with JSBSim showing one of the forces in the LIFT axis and how it's dependent on alpha: 

https://github.com/JSBSim-Team/jsbsim/blob/06f4ce37619d73c619c6296d1f602b10088bd95d/aircraft/737/737.xml#L656-L676

And an example of a moment which is also dependent on alpha:

https://github.com/JSBSim-Team/jsbsim/blob/06f4ce37619d73c619c6296d1f602b10088bd95d/aircraft/737/737.xml#L769-L779

These functions defining the forces and moments are then executed and they are summed in:

https://github.com/JSBSim-Team/jsbsim/blob/06f4ce37619d73c619c6296d1f602b10088bd95d/src/models/FGAerodynamics.cpp#L132

alexandrequ, 2019-03-05:
Thank you @seanmcleod

closed 2019-03-05 by alexandrequ