JSBSim-Team/jsbsim #144: error LNK2019: unresolved external symbol

ganscool86, 2018-12-30:
Hi, 
I am trying to build the JSBSIM source code in Visual Studio 2017 and i am getting the following error:

error LNK2019: unresolved external symbol "public: __thiscall JSBSim::FGTemplateFunc::FGTemplateFunc(class JSBSim::FGFDMExec *,class JSBSim::Element *)" (??0FGTemplateFunc@JSBSim@@QAE@PAVFGFDMExec@1@PAVElement@1@@Z) referenced in function "public: void __thiscall JSBSim::FGFDMExec::AddTemplateFunc(class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const &,class JSBSim::Element *)" (?AddTemplateFunc@FGFDMExec@JSBSim@@QAEXABV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@PAVElement@2@@Z)

Can anyone help in resolving this. Thanks.

seanmcleod, 2018-12-30:
@ganscool86 I've just committed an update to the JSBSim master branch to add the missing reference to `FGTemplateFunc.cpp` to the Visual Studio project file.

ganscool86, 2018-12-30:
Thanks...now able to build successfully.

closed 2018-12-30 by bcoconni