JSBSim-Team/jsbsim #147: JSBSim Scripting Tutorial in linux path issues

josheeg, 2019-01-03:
JSBSim Scripting Tutorial in linux path issues

JSBSim –script=aircraft\Tutorial_1\trim-cruise.xml

The argument "–script" cannot be interpreted as a file name or option.
is what i get back from the terminal. 
where should I run it from and how do i install the tutorials files?

bcoconni, 2019-01-05:
Since you are running JSBSim from Linux I think you should slashes `/` as directory separators rather than backslashes `\`. Therefore the command should be
```bash
> JSBSim –script=aircraft/Tutorial_1/trim-cruise.xml
```

josheeg, 2019-01-05:
JSBSim aircraft/Tutorial_1/scripts/trim-cruise.xml
worked in linux when installed folders merged in ardupilot directory.
textrotate was not found.