JSBSim-Team/jsbsim #152: JSBSim aircraft script tutorial writing output to files xml parse error in linux.

josheeg, 2019-01-06:
JSBSim aircraft/Tutorial_1/scripts/trim-cruise.xml Tutorial_1.csv > JSBSim.out
In file Tutorial_1.csv: line 1
XML parse error: syntax error

the -inputlog -outputlog= .... file is no acsepted in linux

aircraft\Tutorial_1\results\trim-cruise.csv is not there. Tutorial_1.csv is in the home directory from
JSBSim aircraft/Tutorial_1/scripts/trim-cruise.xml

bcoconni, 2019-01-10:
@josheeg 
To be honest, your bug report is a bit confusing, you should tell what you want to get and describe how this is different from what you are actually getting.

I will try to explain some elements based on I understood from your issue report.

According to its name `Tutorial_1.csv` is a [comma separated values file](https://en.wikipedia.org/wiki/Comma-separated_values) which generally contains results/outputs from JSBSim. So I am not sure why you are including it in the command line ?

It might be that you want to rename the file to which results are written in which case the correct command line would be
```bash
> JSBSim aircraft/Tutorial_1/scripts/trim-cruise.xml --outputlogfile=Tutorial_1.csv > JSBSim.out
```
The result of this command line is that the script `aircraft/Tutorial_1/scripts/trim-cruise.xml` will be executed and the computation results will be written in the file `Tutorial_1.csv`

Generally speaking, you can get the options that JSBSim is accepting with the classical command
```bash
> JSBSim --help
```

bcoconni, 2019-02-09:
@josheeg 
Has your problem been fixed ? If not, could you please be more specific about what you are trying to do ?

bcoconni, 2019-02-16:
Connection lost. Issue closed.

closed 2019-02-16 by bcoconni

josheeg, 2019-02-17:
> @josheeg
> To be honest, your bug report is a bit confusing, you should tell what you want to get and describe how this is different from what you are actually getting.
> 
> I will try to explain some elements based on what I understood from your issue report.
> 
> According to its name `Tutorial_1.csv` is a [comma separated values file](https://en.wikipedia.org/wiki/Comma-separated_values) which generally contains results/outputs from JSBSim. So I am not sure why you are including it in the command line ?
> 
> It might be that you want to rename the file to which results are written in which case the correct command line would be
> 
> ```shell
> > JSBSim aircraft/Tutorial_1/scripts/trim-cruise.xml --outputlogfile=Tutorial_1.csv > JSBSim.out
> ```
> 
> The result of this command line is that the script `aircraft/Tutorial_1/scripts/trim-cruise.xml` will be executed and the computed results will be written in the file `Tutorial_1.csv`
> 
> Generally speaking, you can get the options that JSBSim is accepting with the classical command
> 
> ```shell
> > JSBSim --help
> ```

I was fallowing the tutorial to try to understand jsbsim enough to run it in mavproxy like the rascal but the others of jsbsim will not take off and fly a mission. I want to model a aircraft so beyond the rascal what aircraft made in aeromatic++ will fly a mission in mavproxy?