JSBSim-Team/jsbsim #153: Has there been an official release of JSBSim?

Phillip-M-Feldman, 2019-01-09:
When I look at https://github.com/JSBSim-Team/jsbsim/releases, I see the word "pre-release" in an orange box.  It would be nice to have an official release.  Are there any plans for this?

bcoconni, 2019-01-10:
> It would be nice to have an official release. Are there any plans for this?

As I mentioned in the issue #69, the only thing that really is missing is a decision. At the moment, the code is quite stable even though there are currently three issues which are tagged as **bugs** (#31, #89 and #123). Even though issue #31 is not a bug of JSBSim strictly speaking but rather an issue with the design of one the tests of the JSBSim test suite.

My suggestion would be to tag the code as "stable" once the bugs #89 and #123 are fixed.

In the future, we could coordinate JSBSim releases with FlightGear releases (i.e. one every quarter). That way we could collect bugs from FlightGear users in addition to our owns.

@agodemar @seanmcleod 
Does that seem reasonable to you ?

agodemar, 2019-01-10:
> Does that seem reasonable to you ?

@bcoconni yes it is.

Phillip-M-Feldman, 2019-01-11:
#123 is an edge case that is unlikely to affect most users.  If it can be fixed quickly, that's great.  If not, it seems like this should not hold up the release.

bcoconni, 2019-01-12:
@jdmarch 
> Do you have an ETA for a formal release?

All the contributors are volunteers and real life always takes priority. So no ETA as far as I am concerned. The only thing I commit to is that the release of a stable version has taken a very high priority in my JSBSim agenda :smiley:

bcoconni, 2019-01-12:
@Phillip-M-Feldman 
> #123 is an edge case that is unlikely to affect most users. If it can be fixed quickly, that's great. If not, it seems like this should not hold up the release.

The guys developing the [Space Shuttle model](http://wiki.flightgear.org/Space_Shuttle) are most certainly in strong disagreement with your statement :wink:

Phillip-M-Feldman, 2019-02-07:
I see no signs of progress towards a stable release.  Is someone actually working issues #89 and #123?

I offered to try to help with issue #123 if someone can point me to the relevant code, but no one responded to my post.

bcoconni, 2019-02-09:
@Phillip-M-Feldman 
> I see no signs of progress towards a stable release. Is someone actually working issues #89 and #123?
> I offered to [..snip..] , but no one responded to my post.

I think I have already responded to such statements. So quoting myself : 
> All the contributors are volunteers and real life always takes priority.

If that is not good enough for you then I am sorry but as far as I am concerned I am not willing to commit myself to an agenda. I have enough project schedules and deadline pressure in my day job.

bcoconni, 2019-05-01:
I just created a new branch `v1.0.0` to prepare the release of JSBSim 1.0.0
The packages can be downloaded from the [Release Candidate v1.0.0](https://github.com/JSBSim-Team/jsbsim/releases/tag/Release_Candidate_v1.0.0).

Waiting for feedback before making the official release.

bcoconni, 2020-10-19:
As I already mentioned in issue #69, 🎉 **JSBSim 1.1.0 has been released** 🎉
Binaries can be downloaded from the [v1.1.0 release on GitHub](https://github.com/JSBSim-Team/jsbsim/releases/tag/v1.1.0).
Alternatively Python packages can be installed with `pip`:
```bash
> pip install jsbsim
```
Enjoy !

closed 2020-10-19 by bcoconni