JSBSim-Team/jsbsim #162: Gyro Model

rega0051, 2019-01-29:
Gyros indicate rotation rate, not acceleration. Pretty sure it should be using "GetPQRi()". There are also a few references within the file to acceleration that should be changed to either "rate" or "velocity".

https://github.com/JSBSim-Team/jsbsim/blob/253a2663c8799821013df2f79df4a5d386f6cd88/src/models/flight_control/FGGyro.cpp#L76

  //aircraft rates
  vAccel = mT * Propagate->GetPQRi();

bcoconni, 2019-02-09:
@rega0051 
> Pretty sure it should be using "GetPQRi()".

You are correct.

> There are also a few references within the file to acceleration that should be changed to either "rate" or "velocity".

This is likely due to the fact that the code from `FGGyro.cpp` was copied from `FGAccelerometer.cpp` and no one took the time to fix the typos you are reported.

A Pull Request to the project to fix these glitches would be most appreciated.

rega0051, 2019-02-11:
Pull Request submitted #167

bcoconni, 2019-02-11:
I have to say that I did not fully realized what you meant until I saw your patch. You are absolutely correct about the fact that gyros are measuring rotation rates (rad/s) and not rotational accelerations (rad/s^2). In my initial response I was only focusing on the change of the frame with respect to which the measurement was made (was ECEF and is now ECI).

Good catch and thanks for the patch :+1: 

I have just committed your PR to the code. Thanks again for your contribution.

closed 2019-02-16 by rega0051