JSBSim-Team/jsbsim #163: Aircraft Trim Fails Only When Psi = 270 or Psi = -90

omidnabi, 2019-02-04:
Hi, I'm running into a strange issue and I was wondering if anyone else has had this problem before. I am using the tFull mode to trim my aircraft, and the trim always finishes successfully, except when the initial Psi(yaw/heading) is set to either -90 degrees or 270 degrees. If I set the Psi to 270 degrees, the trim passes, however the aircraft loads facing the exact opposite direction (90 degrees). If I set the initial Psi to -90 degrees, the trim fails, and the program exits.

bcoconni, 2019-02-09:
@omidnabi 
Confirmed ! I can reproduce the bug with the script `737_cruise.xml` and setting '<psi>` to 270 deg and -90 deg result in a crash of JSBSim.

I will investigate that.

bcoconni, 2019-02-09:
@omidnabi 
The bug should now be fixed by the commit 4cdc72c.
Let me know if that fixes the problem for you as well.

I have traced this bug back to 2004 ! I can't believe it has remained unnoticed for so many years.
Thanks for the bug report.

omidnabi, 2019-02-09:
Thank you very much! I'll give it a try on Monday.

bcoconni, 2019-03-09:
@omidnabi 
Have you checked if this bug is fixed ? If so, could you please close this issue ?

Thanks.

omidnabi, 2019-03-18:
The fix works great! Thank you guys!

closed 2019-03-18 by omidnabi