JSBSim-Team/jsbsim #169: Free-swiveling gear simulation

it0uchpods, 2019-02-13:
Hi JSB team!
I am asking if it can be added a free-swiveling gear to the JSBsim, so we can simulate these gears which some taildraggers and other planes (G91) use?

Kind Regards,
Josh

bcoconni, 2019-02-15:
> we can simulate these gears which some taildraggers and other planes (G91) use?

Yes, you can get that by specifying the `<castered>` tag in the contact definition such as below
```xml
<contact type="BOGEY" name="TAIL">
    <location unit="IN">
        <x> 239 </x>
        <y> 0 </y>
        <z> -18 </z>
    </location>
    <static_friction> 5 </static_friction>
    <dynamic_friction> 5 </dynamic_friction>
    <rolling_friction> .1 </rolling_friction>
    <spring_coeff unit="LBS/FT"> 200 </spring_coeff>
    <damping_coeff unit="LBS/FT/SEC"> 195 </damping_coeff>
    <castered> 1.0 </castered>
    <brake_group> NONE </brake_group>
    <retractable> 0 </retractable>
</contact>
```

it0uchpods, 2019-02-15:
Oh, so this is already simulated and I did not realize? I will check it out!

Thanks Bertrand!
Josh

closed 2019-02-15 by it0uchpods