JSBSim-Team/jsbsim #170: Bug Report

AnosAhmad, 2019-02-18:
Hello Developer, 

My name is Enas and I am a Ph.D. student. My research focuses on static code analysis. I was looking for open-source software to collect some static code analysis tool warnings. I checked your source code, and I ran one of static code analysis tools over the source code. A number of warnings were generated; some of them may represent true defects in the source code and some may represent false reports, where the tool shows a defect but in reality there is none. 

I need your help by going through the warnings and checking if any of them represent a true defect in your source code. To do that, kindly find the attached file, which contains the warnings. 

Your response and help are highly appreciated. 

Thanks, 
Enas 
 
[warnings.xlsx](https://github.com/JSBSim-Team/jsbsim/files/2873689/warnings.xlsx)

bcoconni, 2020-12-11:
I infer from your files that `strlen` should not be used. However CodeQL does not return that as being a security issue so I'm wondering...
Regarding the problems you found for `FGTrimAnalysis.cpp`, these files are no longer used by JSBSim.

bcoconni, 2024-09-08:
Outdated issue.

closed 2024-09-08 by bcoconni