JSBSim-Team/jsbsim #171: Probably a stupid question about building JSBSim using GCC 4.4.7

omidnabi, 2019-02-20:
Hi,
I have been using an older version of JSBSim that I checked out a while back, and I just decided to upgrade everything to the newest version available on this repo. On my first attempt to build, I'm getting an error on 'nullptr' not being declared in this scope...Does that mean I can't build the newest version without upgrading to a newer compiler?

Thank you!

seanmcleod, 2019-02-21:
You need a compiler that supports C++11.

https://github.com/JSBSim-Team/jsbsim/blob/master/DevelopersDocs.md

According to https://gcc.gnu.org/projects/cxx-status.html 

> GCC 4.8.1 was the first feature-complete implementation of the 2011 C++ standard

bcoconni, 2019-03-09:
> Does that mean I can't build the newest version without upgrading to a newer compiler?

As @seanmcleod told you, you have no choice but to upgrade to a compiler that is compliant with the C++11 specifications.

bcoconni, 2019-03-17:
@omidnabi 
Any update on this topic ?

omidnabi, 2019-03-18:
No, thank you! We will be upgrading our systems soon so we could use the latest version of JSBSim.

closed 2019-03-18 by omidnabi