JSBSim-Team/jsbsim #172: How  to access to the wingspan value in a created function in FGAircraft.cpp

alexandrequ, 2019-03-05:
Hello,
I've created a function in FGAircraft.cpp and I wanted to access to the wingspan, wing area, alpha values. How should I do that ? 
I've tried with in.Wingspan but it didn't worked.
Thanks for your help !

bcoconni, 2019-03-09:
If you've created a new method in `FGAircraft` then wing span and wing area can be accessed from the members `WingSpan` and `WingArea` (no need to dereference them with `in.`)
```c++
  double meanChord = WindArea/WingSpan;
```
Alpha values can be accessed from `FGAuxiliary`
```c++
  FGAuxiliary* Auxiliary = FDMExec->GetAuxiliary();
  double alpha = Auxiliary->Getalpha();
```

bcoconni, 2019-03-17:
@alexandrequ 
Did you succeed to access the wingspan value from your code ?

alexandrequ, 2019-03-23:
Yes thanks

closed 2019-03-23 by alexandrequ