JSBSim-Team/jsbsim #173: AeromatiC++

swampmo, 2019-03-13:
With piston engines I wonder if we could ask how many blades the prop has.

bcoconni, 2020-03-16:
@ermarch
Could you please review this request ? Thanks.

ermarch, 2020-03-17:
The idea has always been to keep the number of parameters low, and adjust the generated configurations afterwards by hand. It's not a replacement for DATCOM and JavaProp. So in that sense   I'm inclined to say no.

But the propeller code needs a good overhaul anyhow since it's output is not the best in the world to say it mildly. I could take the request into account when working on it.

bcoconni, 2020-04-09:
> I could take the request into account when working on it.

Thanks. I've assigned this issue to you and it is now labeled as a `Feature request`.