JSBSim-Team/jsbsim #174: Trim Failure Diagnosis

DBogue, 2019-03-15:
Trim failures for a new configuration are failing with minimal diagnostics. 
Is there a mechanism to log iteration-by-iteration forces/moments and control surface deflections during a trim iteration for diagnostic purposes? 

Is there any documentation for the trim routines?

Thanks!
-David

seanmcleod, 2019-03-16:
I'm not aware of any documentation of the trim routines other than the comments in the source code, which is what I've looked at in the past when I had trim issues. I just double-checked the JSBSim Reference Manual PDF and there isn't any description of the trim algorithm in there.

There is a debug message available for each iteration while solving for each axis, although it's not logging the forces, moments etc.

https://github.com/JSBSim-Team/jsbsim/blob/a10c4385e71e7b0a59025a402996c02efadcaca6/src/initialization/FGTrim.cpp#L556-L581

bcoconni, 2019-03-17:
Unfortunately, the debug messages can only be activated by modifying manually the `Debug` variable in `FGTrim.cpp` and recompiling the source code.
https://github.com/JSBSim-Team/jsbsim/blob/a10c4385e71e7b0a59025a402996c02efadcaca6/src/initialization/FGTrim.cpp#L62-L72
The debug messages will be quite obscure if you don't familiarize yourself with the algorithm first. Basically the trim algorithm is using an iterative, one-axis-at-a-time, scheme.

And there is no documentation except the few comments in the source code. :disappointed: 

I have become familiar with this code when I had to fix bugs in it, so you can post your questions in this issue if you need explanations.

DBogue, 2019-03-17:
Thanks!  Enabling de-bug is first step and source code helps. I was wondering about changing the Newton step size and max iterations.  Will recompile and post response within 24 hrs.

DBogue, 2019-03-19:
Debug messages did the trick.  :)
Thanks for the advice.

closed 2019-03-19 by DBogue