JSBSim-Team/jsbsim #177: Twin engine helicopter model?

litzi2460, 2019-03-19:
Hello,

I wonder if the problem of simulating a twin engine helicopter with JSBSim (see this old post https://sourceforge.net/p/jsbsim/mailman/message/29073718/) has ever been solved, satisfactory?

I am trying to add realistic twin engine parameters like N1, TRQ, N2 for each engine, for display on the VMD, vehicle management display, to a EC145 model in Flightgear (https://forum.flightgear.org/viewtopic.php?f=4&t=24800&p=343195&hilit=ec145#p343195). Currently the fdm in use there is Yasim and the parameters are just "best guesses", produced via Nasal scripts. Maybe I can give it a try with JSBSim, is there a documented solution or example for a plausible JSBSim twin engine helicopter model?

Martin

bcoconni, 2019-03-23:
>  wonder if the problem of simulating a twin engine helicopter with JSBSim (see this old post https://sourceforge.net/p/jsbsim/mailman/message/29073718/) has ever been solved, satisfactory?

No patch has been submitted to address this limitation since the problem was raised in July 2012. I guess the workaround described in the discussion that you have mentioned is still applicable.

The problem is that JSBSim is designed as a one way and one-to-one engine to thruster model.

litzi2460, 2019-03-25:
Thank you for the quick reply!

Meanwhile I have experimented with a workaround that seems to work surprisingly well. I have defined the two engines with dummy rotors  (playing the role of the engines power shaft) and a main and tail rotor with dummy engines.

RPM from the engines (max. of the two) are transferred from the dummy rotors to the main and tail rotors via the external RPM option (<ExternalRPM> -1 </ExternalRPM>). Power demand of the main rotor is in return transfered back to the engines by controlling the dummy rotors brake. To allow this the max. brake power of the dummy rotors has been defined equal to the max. engine power.

To distribute the power demand between the two engines I use a table with an S-curve like function, with input from the dummy rotors (shaft) RPM difference. Thus, in case of RPM difference between the engines, only the faster engine "drives" the rotor.

The challenge I currently face is to design a "neutral" dummy rotor (for simulating the shaft) with minimum negative side effect on the overall models flight characteristics.

You may close the issue, unless you might want me to post a final solution, later on, to this issue.

jd60fe, 2019-06-10:
litzi,

I would be very interested in your final product and would like to see a copy of it.  I am working on a similar issue for the UH-1N since it uses a combining gearbox but I need to model the N1 and N2 or the two engines input into the CGB.

jd

litzi2460, 2019-06-14:
Hi,

I have provided the relevant files for the Flightgear Alouette-III model with the twin engine modification, as an example in the link below:

https://www.dropbox.com/s/vddzv709jpnwl5o/Alouette-III-jsbsim-twin-engine.zip?dl=0

This is still very experimental, issues may occur in any flight phases. Especially to mention are unresolved things : Oscillation of N2 during the syncronisation phase (during spoolup) of the engines, making a dummy rotor with minimum negative side effect on the overall models flight characteristics.

Martin

jd60fe, 2019-06-14:
Thank you sir.