JSBSim-Team/jsbsim #178: FGTurbine: Windmilling disable option?

it0uchpods, 2019-03-21:
Hi guys,
I am thinking about making an APU in JSBsim engine, as apposed to Nasal.
The inlet for the APU in the MD-11 (700-4E) is placed in a way that no windmilling would occur in flight.

Is it possible to add an option to disable windmilling?

Thank you!
Kind Regards,
Josh

bcoconni, 2019-03-23:
`FGTurbine` is not a detailed engineering model of turbine engines: it does not simulate the physics of the engine but rather imitates the behavior of a generic turbine engine.

Therefore windmilling is not accounted for in `FGTurbine` and you do not need to disable it.

it0uchpods, 2019-03-23:
Hi @bcoconni I do not think this is true:
Low Speed: https://cdn.discordapp.com/attachments/541330527106498562/559149238991454238/Screenshot_2.png

High Speed: https://cdn.discordapp.com/attachments/541330527106498562/559149239125540875/Screenshot_1.png

What is strange, at high speed the offline N1 is much higher than should be... 38%?????

These values are right from FG's /engines/engine[x]/n1 (or n2) props.

Kind Regards,
Josh

NikolaiVChr, 2019-03-24:
Seems to me it is simulated somehow..

When I run out of fuel, N1 will still spin, if I dive down N1 goes up as I pick up speed. Is that not windmilling?

bcoconni, 2019-03-24:
@it0uchpods What was the aircraft you used when you took the pictures ?

Are the properties `/engines/engine[*]/n1` and `/fdm/jsbsim/propulsion/engine[*]/n1` in agreement during windmilling ? If that is not the case then the windmilling is simulated by JSBSim but most likely by some Nasal script.

And BTW, are the properties `/fdm/jsbsim/propulsion/engine[x]/thrust-lbs` negative during windmilling ? I would be very, very surprised that this occurs because I did not find a single line of code in `FGTurbine.cpp` that produces negative thrust. And this why I claimed that windmilling **is not** simulated in JSBSim turbine engine model.
https://github.com/JSBSim-Team/jsbsim/blob/114173fdc42a25dd515596084b495e9976857eda/src/models/propulsion/FGTurbine.cpp#L1-L666

> What is strange, at high speed the offline N1 is much higher than should be... 38%?????

I guess it depends on a number of factors. Of course it depends on the aircraft velocity but it also depends on the altitude because the higher the air density is, the higher is the energy transmitted by the air to the fan (N1). However, since the pictures show a glass cockpit, I guess the aircraft is equipped with modern high bypass ratio engines and indeed for such engines it seems unlikely to reach 38%N1 while windmilling. Your best option would be to check the engine documentation if you have an access to it.

25%N2 seems closer to reality because at low to medium altitudes windmilling should be able to bring your engine HP rotation speed to a value close to what the starter reaches and this allows to relight your engine.

it0uchpods, 2019-03-24:
I am using the IDG-MD-11X, (my github)
I wrote all the nasal in this aircraft from scratch, and I know for sure there is no such nasal aircraft side.

JSB's props match fully: https://cdn.discordapp.com/attachments/559385110332309564/559385152153976852/Screenshot_1.png

Kind Regards,
Josh
PS: Disregard the lack of EPR param in this time, I am using GE engines instead of PW.

bcoconni, 2019-04-07:
@it0uchpods 
The thrust is 0 as I was anticipating but the rotation speeds N1 and N2 are non zero which is most interesting. What was the throttle position when you captured your screen shots ?

it0uchpods, 2019-04-07:
Doesn't make a difference, but in this case if it will have been probably somewhere around half guessing from the N1 command (The T on top of the N1 dials)

Kind Regards,
Josh

bcoconni, 2019-04-07:
OK so I finally found it.
https://github.com/JSBSim-Team/jsbsim/blob/7bb04beb8516a874ad81914dd1a0f62b6d238ca3/src/models/propulsion/FGTurbine.cpp#L186-L191

`N1` and `N2` are hard coded to be respectively equal to `qbar/10` and `qbar/15` when the engine is off. The mass airflow would more relevant than the dynamic pressure for that matter but never mind.

So indeed at the moment, it is not possible to disable this "windmilling". But does it really matter since it does not produce any physical effect (i.e. drag) and only makes up numbers ?

it0uchpods, 2019-04-07:
Would it be possible to Add such a disable anyways? Otherwise I have to fake my N1/N2 numbers for the APU (right now I just lag them), and perhaps in the future it will be detailed. In my book, still a good idea.

Kind Regards,
Josh

bcoconni, 2019-05-08:
I just pushed the patch to FlightGear `next` branch.