JSBSim-Team/jsbsim #180: How to make a simple flight of a c172x

alexandrequ, 2019-04-09:
hello,
I'm trying to make a simple flight of a C172x.
First of all I use a file "airborne" where I set a velocity of 55 m/s and a position in the ECEF frame of (6373000,0,0) in meters.

I would like to make the aircraft simply flight in the zECEF direction and in a second time I would make the aircraft fly in a wake. 
For that I use the following file : "SimpleFly" for a simple flying path in zECEF 
But the aircraft isn't stable at all. Why ? And what should I do ? 
Thank for the help !

The airborne.xml 
```
<?xml version="1.0" encoding="utf-8"?>
<initialize name="airborne">

  <!--
     This file sets up the aircraft initial conditions. It turns
	 all of the engines on, and provides a default altitude and
	 airspeed. This may be overridden in the script file that
	 calls this initialization file.
  -->
  <running>                -1   </running>
  <!-- <altitude unit="FT">  5000.0  </altitude> -->
  <vc unit="M/S">        55.0  </vc>
  <position frame="ECEF">
      <x unit="M"> 6373000 </x>
      <y unit="M">        0 </y>
      <z unit="M">        0 </z>
    </position>
</initialize>
```

And the SimpleFly.xml code : 

```
<?xml version="1.0" encoding="utf-8"?>
<?xml-stylesheet type="text/xsl" href="http://jsbsim.sf.net/JSBSimScript.xsl"?>
<runscript xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance"
    xsi:noNamespaceSchemaLocation="http://jsbsim.sf.net/JSBSimScript.xsd"
    name="Trim - Approach">

    <description>
        Trimmed at altitude/airspeed in Approach configuration,
        run for 4 second

        Author:
        Bill Galbraith
        Holy Cows, Inc.
        billg@holycows.net
        November 2010
    </description>

    <use aircraft="c172x" initialize="airborne"/> <!--pa28-->
    <run start="0" end="40" dt="0.00833333">
  </run>

  <output name="roll.csv" type="CSV" rate="120">
      <rates>            ON </rates>
      <velocities>       ON </velocities>
      <position>         ON </position>
      <atmosphere>       ON </atmosphere>
      <fcs>              OFF </fcs>
      <ground_reactions> OFF </ground_reactions>
      <propulsion>       OFF </propulsion>
      <simulation>       OFF </simulation>
      <massprops>        OFF </massprops>
      <forces>           ON </forces>
      <moments>          ON </moments>
      <aerosurfaces>     OFF </aerosurfaces>
      <coefficients>     ON </coefficients>

  </output>
</runscript>

```

bcoconni, 2019-04-14:
In `airborne.xml`, you are using features (`<position frame="ECEF">`) that are only available for version 2.0 of initialization files. So you must add the attribute `version="2.0"` to the `<initialize>` tag of `airborne.xml`:
```xml
<?xml version="1.0" encoding="utf-8"?>
<initialize name="airborne" version="2.0">
```
However the `<vc>` tag is ignored in version 2.0 of initialization files so the aircraft will start with a zero velocity.

In addition, units cannot be specified individually for each parameter `x`, `y` and `z`. Instead it must specified in the `<position>` tag:
```xml
  <position frame="ECEF" unit="M">
    <x> 6373000 </x>
    <y>       0 </y>
    <z>       0 </z>
  </position>
```

Finally, given your choice of `x`, `y` and `z` position, the aircraft starts at the equator (latitude=0°). Since JSBSim uses the WGS84 Earth parameters, the Earth radius at the equator is approximately 6378km meaning that your aircraft spawns 5 km below ground level.

My suggestion to start the aircraft 5000ft above ground level with a calibrated speed of 55m/s heading North (hence flying along zECEF):
```xml
<?xml version="1.0" encoding="utf-8"?>
<initialize name="airborne">
  <running>                -1   </running>
  <altitude unit="FT">  5000.0  </altitude>
  <vc unit="M/S">        55.0  </vc>
</initialize>
```
By default, JSBSim initialize the aircraft position at latitude=0° and longitude=0° heading North so you do not need to specify these parameters.

alexandrequ, 2019-04-18:
Thanks @bcoconni  !
And now I want to try to maintain that direction even if my aircraft meet a wake of an other aircraft (that I simulate and have add to JSBSim), for the moment this is what's happening : [here](https://drive.google.com/drive/folders/1udZxdKsBbt_7ZwDOPnxY7cNYmqr_tCPl?usp=sharing) 
Is there a autopilot that I could call to try to maintain a straight path ? If yes, how should I call it ? 
Thanks for helping me.

bcoconni, 2019-04-20:
> How could I define a direction who would be an an angle of 45° from the zECEF direction ?

I guess you mean 45° from North ? In that case use the parameter `<psi>` to define the heading in your init file
```xml
<?xml version="1.0" encoding="utf-8"?>
<initialize name="airborne">
  <running>            -1     </running>
  <psi unit="DEG">     45.0   </psi>
  <altitude unit="FT"> 5000.0 </altitude>
  <vc unit="M/S">      55.0   </vc>
</initialize>
```
> Is there a autopilot that I could call to try to maintain a straight path like a track hold autopilot (image) ? If yes, how should I call it ?

There is an autopilot with the c172. See the script `c172_cruise_8K.xml`. In general there are 2 steps, first, trim the aircraft and then set up the autopilot (in the example below the aircraft is set to maintain 4000 ft heading 200°).
https://github.com/JSBSim-Team/jsbsim/blob/9755424f1021d618ab8e150afd849316b4070f86/scripts/c172_cruise_8K.xml#L80-L87
It is a general purpose autopilot so you might need to tweak the autopilot parameters for the aircraft to keep its heading/altitude in a strong wake.

alexandrequ, 2019-04-25:
Thank for the answer !
But I don't understand why I don't see any reaction of the aircraft guidance elements when the autopilot is set on ON with c172_cruise_8K.xml ? 
[image]

bcoconni, 2019-04-27:
> But I don't understand why I don't see any reaction of the aircraft guidance elements when the autopilot is set on ON with `c172_cruise_8K.xml` ?

Because the aircraft is trimmed with a heading of 200° and an altitude of 4000ft
https://github.com/JSBSim-Team/jsbsim/blob/6190fcc32051fd96c601b329ed320a67cacdc725/aircraft/c172x/reset01.xml#L8-L9
and the autopilot is set to maintain the same heading (200°) with the same altitude (4000ft).
https://github.com/JSBSim-Team/jsbsim/blob/6190fcc32051fd96c601b329ed320a67cacdc725/scripts/c172_cruise_8K.xml#L82-L85
The flat curves therefore indicate that the aircraft trimming was of such a good quality that the autopilot does not need to update the controls to maintain the heading and altitude.

If you want to get some funny results, set the autopilot parameters in `c172_cruise_8K.xml` lines 82 and 84 to values different from the initial conditions from `reset01.xml`.

alexandrequ, 2019-05-08:
Hello, I've tried the autopilot but it doesn't seem to work. I applied a wind on the aircraft but anything happens. Do you have an idea why ? 

[image]
 
[image]

When I add 
`  <set name="ap/roll-attitude-mode" value="0"/>  `
  `    <set name="ap/autopilot-roll-on" value="1"/>`

Then the rudder moves but with only 4°,  it's seems to be quite low for a c172 flying in the wake of a B747. What do you think about it ?
[image]

alexandrequ, 2019-05-16:
Hello,
I'm trying the heading_hold autopilot for the c172 and it doesn't do anything. Do you have an idea why ? 
``
    <set name="ap/heading_setpoint" value="90"/>
    <set name="ap/heading_hold" value="1"/>
``

[image]
[image]

bcoconni, 2019-05-18:
> I'm trying the heading_hold autopilot for the c172 and it doesn't do anything.

Indeed. For some reason the commit 0c0367e7b3bda25a0c1781ac877993bab69e00e3 commented out a line of code in the heading autopilot that now makes it inoperant. You should restore the feature with the following modification to the file `aircraft/c172x/c172x.xml`:
```diff
--- a/aircraft/c172x/c172x.xml
+++ b/aircraft/c172x/c172x.xml
@@ -370,7 +370,7 @@
         <channel name="Roll">
 
             <summer name="fcs/roll-trim-sum">
-<!--                <input>ap/aileron_cmd</input> -->
+                <input>ap/aileron_cmd</input>
                 <input>ap/roll-cmd-norm-output</input>
                 <input>fcs/aileron-cmd-norm</input>
                 <input>fcs/roll-trim-cmd-norm</input>
```

alexandrequ, 2019-05-23:
Thanks for all !  It's working now.

closed 2019-05-23 by alexandrequ