JSBSim-Team/jsbsim #181: Regression: switch statement no longer allows use of and

Zaretto, 2019-04-13:
Probably as a result of the change to support nested tests in the switch statement the use of the and/or elements no longer works.

This example switch condition the test value="1" never becomes true even when all of the 
conditions are satisfied.

```
       <switch name="systems/cadc/manuv-active">
         <default value="0.0"/>

         <test value="0">
           <or>
             fcs/flap-cmd-norm gt 0.3714285714285714
             aero/alpha-deg lt systems/cadc/manuv-flapslat-retraction-aoa
             velocities/mach gt 
systems/cadc/manuv-flap-partially-extended-max-mn
             gear/gear-cmd-norm ne 0
           </or>
           systems/cadc/manuv-active ne 0
         </test>

         <test value="1">
           <or>
             <and>
               fcs/flap-cmd-norm le 0.3714285714285714
               aero/alpha-deg gt systems/cadc/manuv-flapslat-extension-aoa
               velocities/mach le 
systems/cadc/manuv-flap-fully-extended-max-mn
               gear/gear-cmd-norm eq 0
               fcs/wing-sweep-cmd le 0.7352941176470588
             </and>
             systems/cadc/manuv-active ne 0
           </or>
         </test>
       </switch>

```
The above switch statement has been tested and is working on 2018.3 and a 2018.4 nightly prior to the change.

If I change the logic to use embedded <test logic="OR|AND"> then the 
switch works; however I suspect that this may well break a few existing models and retaining the ability of JSBSim to continue to provide backwards compatibility is important.

bcoconni, 2019-04-14:
I am afraid the keywords/tags `<and>` and `<or>` have never been supported by JSBSim.

I dug JSBSim code history and never found a mention of these 2 keywords being read and interpreted as a replacement for `<test logic="{AND|OR}">`.

However, JSBSim have always had a relaxed interpretation of the tags within `<switch>` and `<condition>` statements and that allows tags `<and>` to be interpreted as you meant them but it is merely a coincidence.

Actually **any** tag within a `<test>` statement is interpreted as `<test logic="AND">`: you could have used `<smuglob>` instead of `<and>` and JSBSim would have interpreted `<smuglob>` as `<test logic="AND">` nonetheless.

JSBSim only checks for the `logic` attribute so unless you specify `<or logic="OR">` the tag `<or>` will be interpreted as `<test logic="AND">`.

After a few tests that I conducted, I am pretty much convinced that JSBSim interpreted your code differently than what you meant. In the code below the tags `<and>` and `<or>` have been replaced by `<test logic="AND">` and the ignored statement have been commented out `` to illustrate how JSBSim actually interpreted your code:
```xml
<switch name="systems/cadc/manuv-active">
  <default value="0.0"/>

  <test value="0">
    <test logic="AND">
      fcs/flap-cmd-norm gt 0.3714285714285714
      aero/alpha-deg lt systems/cadc/manuv-flapslat-retraction-aoa
      velocities/mach gt systems/cadc/manuv-flap-partially-extended-max-mn
      gear/gear-cmd-norm ne 0
    </test>
    <!-- Ignored statement: systems/cadc/manuv-active ne 0 -->
  </test>

  <test value="1">
    <test logic="AND">
      <test logic="AND">
        fcs/flap-cmd-norm le 0.3714285714285714
        aero/alpha-deg gt systems/cadc/manuv-flapslat-extension-aoa
        velocities/mach le systems/cadc/manuv-flap-fully-extended-max-mn
        gear/gear-cmd-norm eq 0
        fcs/wing-sweep-cmd le 0.7352941176470588
      </test>
    </test>
    <!-- Ignore statement: systems/cadc/manuv-active ne 0 -->
    </test>
</switch>
```
I would be interested if you find a different behavior according to your tests.

## Way forward ##
If you confirm that JSBSim has always misinterpreted the code as described above, then unfortunately I do not see a way forward that would save models adaptations.
### Option 1: Make JSBSim perform stricter syntax checking and reject illegal tags including `<and>` and `<or>` ###
That would put an end to the use of illegal tags (including `<and>` and `<or>`) and that would force all the models authors to fix their code which would be kind of a breach to backward compatibility.
### Option 2 : Make the tags `<and>` and `<or>` legal ###
Given that JSBSim historically interpreted the `<or>` tag as an `<and>` tag that would modify the output of `<switch>` statements that were using `<or>` and that would also break some existing models.
### Option 3 : leave the code as is ###
I don't think this is a reasonable option but I mentioned it for the sake of completeness.

Options 1 and 2 are quite easy to add to JSBSim but I am unsure which one would be the best or not so bad...

Zaretto, 2019-04-17:
Thanks for clarifying the issue; your explanation makes perfect sense of the situation that was previously somewhat confusing for me.

Testing showed that your revised code operated as I'd expect it to; which proves 

Maintaining compatibility is a tricky question for this; my thoughts are;

1. Given that the current JSBSim already performs more strict syntax checking option(1) is the safest option; although it will break undoubtedly break some models this is preferable to silent failure or inconsistent results.

2. Option 2 is tempting; however as you said this could cause working models to behave differently and therefore is not desirable 

3, As you said option 3 is not really viable.

4. I'm proposing a fourth option which is to implement option(2) but with console warnings that the behaviour may have changed. 

After much consideration I think my option(4) above is my preferred choice; even though the behaviour may be changed it is likely to be what the modeler intended; the main reason I prefer this over option(1) is that it is more consistent with the use of the <and> and <or> operations in <ifthen>

However option(1) is my second choice option; it's the safest option and generally tighter syntax checking of the XML can only be a good thing.

bcoconni, 2019-04-20:
I understand your point but I am not sure that going straight to option 4 is a good option : I'm afraid that the warning messages will go unnoticed while the aircraft behavior will be affected silently.

I would rather go for a 2 stages process:
1. I will implement option 1 to force the model designers to fix all the abusive `<or>` and `<and>` tags in their models.
2. The keywords `<and>` and `<or>` will be added for convenience.

Stage 2 would have to wait for at least one or two releases of FlightGear in order to make sure there remain no spurious `<and>` and `<or>` tag in the JSBSim models. Also I think that such a radical change needs to be discussed with the FlightGear folks so we need to discuss that on their mailing list.

Zaretto, 2019-04-21:
It's definitely something that is worth discussing on the FG list; however we already have model breaking changes going into the next version of FG (related to the fatal exceptions that you can now get from bad functions (i.e. missing parameters))- so in that sense it'd be good to get all breaking changes into the next FG and we can fix the bad models. I actually suspect that the vast majority of models will be OK - as a lot of the bad XML is only on a few craft (mine included).

So overall what you've done is a good solution and I'm happy with that; my only suggestion is that we discuss making the errors on the abusive and/or to be fatal exceptions rather than just console output.

bcoconni, 2019-04-27:
Option 1 is now available in FlightGear `next` branch. As a fallback option, I have also made option 4 available as a [merge request (158)](https://sourceforge.net/p/flightgear/flightgear/merge-requests/158) to FlightGear.

bcoconni, 2019-07-13:
@Zaretto 
Should we consider this issue closed or do you need further support ?

Zaretto, 2019-07-14:
Issue closed as now resolved.

closed 2019-07-14 by Zaretto