JSBSim-Team/jsbsim #182: Parsing behaviour of extraneous <property> inside an auto value/property element

Zaretto, 2019-04-17:
With the current changes to the XML parser the following (invalid) statement no longer works; I suspect that previously the `<property>` tags on the `max` operator were simply being ignored.

```      <clipto>
        <min> 0 </min>
        <max><property>systems/hydraulics/normal-pressure</property></max>
      </clipto>
```
what I really meant of course is the following;
```     <clipto>
        <min> 0 </min>
        <max>systems/hydraulics/normal-pressure</max>
      </clipto>
```
Can we make this a syntax validation failure please.

bcoconni, 2019-04-20:
This has been added in commit 61030f0.

bcoconni, 2019-04-27:
And this is now available in FlightGear `next` branch.

bcoconni, 2019-05-18:
@Zaretto 
Have you had the opportunity to test that feature in FlightGear ?

Zaretto, 2019-05-18:
I've tested it and it worked fine.

Thanks for doing this.

closed 2019-05-18 by bcoconni