JSBSim-Team/jsbsim #186: Transmit simulation time via FG UDP interface

simfinite, 2019-04-28:
Hey all,

over at ArduPilot/ArduPlane [we are having some trouble](https://github.com/ArduPilot/ardupilot/issues/10612) to reliably synchronize our autopilot simulation application to a JSBSim instance using the FG UDP interface. We would benefit a lot from having the JSBSim's internal simulation time transmitted via the interface along with the FDM state vector.

I noticed a member `cur_time` in the `FGNetFDM` struct that is being sent over this interface. This member is currently documented to hold the unix time but it is unused and set to a 'dummy' value of `1234567890`.  I would like to propose using this existing member to transmit the simulation time. The required changes [be seen in this commit](https://github.com/simfinite/jsbsim/commit/f91283583fabac2b271ce0f84e6484164c37e15a).

If you agree to make this change, I will prepare a pull-request with the required changes in FGNetFDM.hxx and FGOutputFG.cpp. If not, then we can discuss other possible ways of having the simulation time transmitted.

bcoconni, 2019-04-28:
> I noticed a member cur_time in the FGNetFDM struct that is being sent over this interface. This member is currently documented to hold the unix time but it is unused and set to a 'dummy' value of 1234567890. I would like to propose using this existing member to transmit the simulation time.

Makes sense to me.
> If you agree to make this change, I will prepare a pull-request with the required changes in `FGNetFDM.hxx` and `FGOutputFG.cpp`. 

I guess you mean `net_fdm.hxx` instead of `FGNetFDM.hxx` ? Anyway, I don't see any change on either file in the commit you pointed to. Be careful about the fact that the content of `net_fdm.hxx` is managed by the FlightGear project and that we cannot make any alteration to it before synchronizing with them.

simfinite, 2019-04-28:
> I guess you mean net_fdm.hxx

Yes. The documenting comment for the member `cur_time` should be changed, see the pull request for my proposal. Since it's "just a comment", I would hope we can go forward with it, but I don't know the procedure.

bcoconni, 2019-05-09:
Your PR has just been merged.
Thanks for your contribution.

closed 2019-05-09 by bcoconni

simfinite, 2019-05-10:
Thank you for the review and help with this.