JSBSim-Team/jsbsim #189: python tests argv[1]

3Kvil, 2019-05-09:
When I tried to run test case in my Pycharm(Win10 Python3.7), it had a error

 File "D:\study\datcom_\JSBSIM\jsbsim\tests\JSBSim_utils.py", line 142, in setUp
    self.sandbox = SandBox(*args)
  File "D:\study\datcom_\JSBSIM\jsbsim\tests\JSBSim_utils.py", line 30, in __init__
    path_to_jsbsim = os.path.join(sys.argv[1], *args)
IndexError: list index out of range

Should I run the test with a extra argv passed to the test script?
how should I start a simple python program which could generate some results.
PS: I am recently trying to use JSBSim' s python module for flight performance evaluation. Confused by the logic of JSBSim software framework.

bcoconni, 2019-05-09:
The test cases are not meant to be run from the Python interpreter. Instead you should use the Python script `JSBSim` that should have been installed (I am assuming you used `pip install` to install JSBSim Python modules and script.).

So if you want to execute the script `c1721.xml`, just run
```bash
> JSBSim --script=scripts/c1721.xml
```
For more options, the usual option `--help` can be used
```bash
> JSBSim --help
```

3Kvil, 2019-05-10:
JSBSim --script=scripts/c1721.xml runs good on my Windows Powershell,THX! Does it run a C++ compiled .exe program or a Python script that uses a  Cython .pyd module?

bcoconni, 2019-05-18:
> Does it run a C++ compiled .exe program or a Python script that uses a Cython .pyd module?

Well, it mostly depends if you ran `pip install` to install JSBSim or not.
- If you are running on Linux or Mac you can check where the script is located with the terminal command `which JSBSim` and infer if it is the Python script or not.
- If you are running on Windows then the C++ executable name should have the `.exe` extension while the Python script has not.

3Kvil, 2019-06-06:
So. can we use a simple os module to call the C++ .exe
`import os
os.system('JSBSim --script=scripts/737_cruise_steady_turn.xml')`
to replace the following code:
`    fdm = jsbsim.FGFDMExec(root_dir='../../')
    print(fdm)
    print(fdm['atmosphere/rho-slugs_ft3'])

    currentdir = os.getcwd()
    scriptsdir_path = '../../scripts'
    scripts = []
    for f in os.listdir(scriptsdir_path):
        script_path = os.path.join(scriptsdir_path, f)
        # Does f contains a JSBSim script ?
        if CheckXMLFile(script_path, 'runscript'):
            # print(script_path)
            scripts.append(script_path[len('../../'):])
    print(scripts)
    fdm.load_script(scripts[1]) 
    fdm.run()`
What is the difference?
Also, JSBSim.exe can a transmit flight dynamics data through UDP, can the jsbsim module does the same?

bcoconni, 2019-06-08:
The big chunk of Python code you show is running **all** scripts in a row while the pair of code lines at the top is just executing the script `737_cruise_steady_turn.xml`.

The minimum amout of Python code is
```python
import jsbsim

fdm = jsbsim.FGFDMExec(root_dir='../..')
fdm.load_script('scripts/737_cruise_steady_turn.xml')
while fdm.run():
  pass
```
The Python script `python/JSBSim.py` is doing that for you and provides almost the same features than `JSBSim.exe` (it is still work in progress)
https://github.com/JSBSim-Team/jsbsim/blob/8a0cff17c905fd18007180859656322c4998eee5/python/JSBSim.py#L1-L113

> Also, JSBSim.exe can a transmit flight dynamics data through UDP, can the jsbsim module does the same?

Yes, the JSBSim Python module has exactly the same features than `JSBSim.exe` for the simple reason that the Python module is just a thin layer of Python to C interface code (i.e. a wrapper) over the C++ code that is used for `JSBSim.exe`.

bcoconni, 2019-07-13:
@3Kvil 
Have all your questions been answered or do you need further support ?

3Kvil, 2019-07-15:
Yes, thank you for your patient reponse! I have successfully developed the code for interact with JSBSim  to run the .exe program and receive data using UDP using multi-threading. But I haven't succeeded in revceiving the data from the python module of JSBSim. Will continue to test  that. Next step would be trying to connect the flightgear using the python module.

bcoconni, 2019-07-15:
> Will continue to test that. Next step would be trying to connect the flightgear using the python module.

No problem. This issue will be closed. Please open a new issue if you need any further support.

closed 2019-07-15 by bcoconni