JSBSim-Team/jsbsim #191: Trim a Rascal 110

alexandrequ, 2019-05-24:
Hello,
I'm actually trying to trim a Rascal 110 UAV model but it doesn't work. 
I've downloaded the model here :  [http://wiki.flightgear.org/Rascal_110](http://wiki.flightgear.org/Rascal_110)
And I initialise it : 
``
<initialize name="Inizialization for Tutorial 01">
  <vt unit="M/S">         30  </vt>
  <latitude unit="DEG">   0.0  </latitude>
  <longitude unit="DEG">  0.0  </longitude>
  <psi unit="DEG">       0.0  </psi>
  <altitude unit="M">  50.0  </altitude>
  <gamma unit="DEG">       0.0  </gamma>
  <running>                -1   </running>
</initialize>
``

Is it because the model isn't fiable or my initial inputs wrong ?
Thanks for help !

alexandrequ, 2019-05-28:
The problem seems to be that the throttle is equal to 0.5

bcoconni, 2019-06-05:
> The problem seems to be that the throttle is equal to 0.5

This generally occurs when the engine is not turned on at the moment the trim is run.

bcoconni, 2019-08-15:
@alexandrequ Any progress on this issue ?

bcoconni, 2019-08-24:
Contact lost. Issue closed.

closed 2019-08-24 by bcoconni

alexandrequ, 2019-08-25:
Hello @bcoconni, sorry for not answering, I was in vacation. I have solved this problem by recreating the model of the Rascal 110 with Aeromatic. 
Thanks for your help.

linuxguy123, 2019-12-16:
For those that might be reading this thread, it might have been related to the faulty parameters in the Rascal.xml file mentioned in this thread. https://github.com/JSBSim-Team/jsbsim/issues/63