JSBSim-Team/jsbsim #192: Qestion about modeling a missile launch

omidnabi, 2019-06-03:
Hi,
This is not exactly an issue, but I was wondering if JSBSim could model missile launch. I have a JSBSim aircraft model, and a separate missile model. I was wondering if I could model the missile as a point mass, and the delete it at launch and let the flight controller take care of changes in C.G. Would this be possible?

Thank you!

bcoconni, 2019-06-05:
>  I was wondering if I could model the missile as a point mass, and the delete it at launch and let the flight controller take care of changes in C.G. Would this be possible?

It is indeed possible. This is exactly the procedure that is followed for rocket stage separation. See the script `scripts/J2641.xml` for instance https://github.com/JSBSim-Team/jsbsim/blob/8a0cff17c905fd18007180859656322c4998eee5/scripts/J2461.xml#L132-L140

omidnabi, 2019-06-05:
Wonderful! Thank you so much!

If I already have an autopilot do I need to make changes to it for
stability or does the program just calculates the new CG and I could just
use the same autopilot with the same gains?

bcoconni, 2019-06-08:
> If I already have an autopilot do I need to make changes to it for stability or does the program just calculates the new CG and I could just use the same autopilot with the same gains?

Your best option is most likely to just try and see how your autopilot behaves. Then you might need to make adjustments. What I call tell for sure is that the physics of the CG shift resulting from the point mass nullification is correctly managed by JSBSim.

Sorry to give such a vague answer but you are not providing much details so it is difficult to give you a detailed answer in return.

omidnabi, 2019-06-13:
Great, thank you so much!!

closed 2019-06-13 by omidnabi

mrwangyou, 2022-07-09:
@omidnabi Hi. I'm trying to work on the situation of confrontation between planes. I search missile models on google but can't find any. I wonder if your missile model is publicly available and where I can get them? 👍 Thank you.

seanmcleod, 2022-07-09:
@mrwangyou you do realise that the "missile" you're referring to here isn't an air to air missile but rather a rocket with an aircraft attached, think of a rocket carrying something like a space shuttle.

mrwangyou, 2022-07-11:
> @mrwangyou you do realise that the "missile" you're referring to here isn't an air to air missile but rather a rocket with an aircraft attached, think of a rocket carrying something like a space shuttle.

Got it. I'll consider it. Moreover, I think maybe `aircraft/J246` is a type of what you said "a rocket with an aircraft attached".