JSBSim-Team/jsbsim #194: [Question] Checkpointing JSBSim

galleon, 2019-06-15:
Greetings from LFBO !

Thanks for your excellent work!

I am looking at saving the state of a simulation on disk and restart it later.

Is there a way to do that in the C library (or better in Python) ?
If not as someone already experienced about tools like dmctp (http://dmtcp.sourceforge.net)?

side question: any doc on the Python interface ?

Cheers
Guillaume

bcoconni, 2019-07-02:
> I am looking at saving the state of a simulation on disk and restart it later.
> Is there a way to do that in the C library (or better in Python) ?

There is currently the option to export the current aircraft state in an XML file to be used as initial conditions for later use.

However after having visited the link you provided I think you are looking for something much more complex than the feature above in which case the answer is currently no, JSBSim does not provide such a feature.

> If not as someone already experienced about tools like dmctp (http://dmtcp.sourceforge.net)?

I personally haven't.
> side question: any doc on the Python interface ?

Unfortunately, no. The wrap up of the C++ interface in Python is currently work in progress and is far from complete. Your best option at the moment is to use the [docs for the C++ API](https://jsbsim-team.github.io/jsbsim/) and the Python interface as is described in the file [python/jsbsim.pyx](https://github.com/JSBSim-Team/jsbsim/blob/master/python/jsbsim.pyx)

bcoconni, 2019-08-15:
@galleon Do you need any further support on this topic ?

bcoconni, 2019-09-01:
FYI the Python module API is now documented and can be accessed from https://jsbsim-team.github.io/jsbsim/python/index.html

galleon, 2019-11-12:
Hi Bertrand,
Thanks for the link to the Python documentation
Sorry for the long silence. Actually I still need this capability. We have tried to implement it ourselves but are facing some discrepancies when trying to re-run a simulation from a previous state. Some properties seem to be not correctly initialized.

You mentionned that there is currently the option to export the current aircraft state in an XML file so that it can be used as initial conditions for later use. Can you point me to that function ? Does it has a Python interface ?

galleon, 2019-11-12:
@bcoconni can I re-open the issue ?

bcoconni, 2019-11-12:
> @bcoconni can I re-open the issue ?

Sure ! No problem.

> You mentionned that there is currently the option to export the current aircraft state in an XML file so that it can be used as initial conditions for later use. Can you point me to that function ? Does it has a Python interface ?

Right. This feature allows to generate an XML file containing initial conditions from which the simulation can be restarted later. Typically the generated file is similar to the example just below https://github.com/JSBSim-Team/jsbsim/blob/82ea6c67b1b9dbb3139e8e1c1323c6a2158c6ae1/aircraft/c172x/reset00.xml#L1-L18
However after a (very) quick glance at your code I understand that you are looking for a much deeper copy of the property tree ?

galleon, 2019-11-13:
Thanks for the quick answer.
Yes what we want is to store the internal state of JSBSim after a few time step so that we can restart the simulation from the same point and see what other pilot steering commands would do.
We are not even sure that this can be done by storing and reading back all properties since we are wondering if there are no others hidden (i.e. not stored) states. What is your point of view of the feasibility?

Sent from a small device

bcoconni, 2019-11-16:
> what we want is to store the internal state of JSBSim after a few time step so that we can restart the simulation from the same point and see what other pilot steering commands would do.
We are not even sure that this can be done by storing and reading back all properties since we are wondering if there are no others hidden (i.e. not stored) states. What is your point of view of the feasibility?

I'd say it is feasible, at least in theory. However some properties are read-only while some others are write-only so I am not sure that the code below (extracted from your project) gives the expected result since the write-only properties will return garbage.
```python
def get_sim_state(self):
    return {prop: self.get_property_value(prop) for prop in Catalog.values()}
```
In addition, if you are restarting JSBSim using the property `simulation/reset` then some of the properties are overwritten by JSBSim in the process so you need to know what you are doing in order to get the expected result. And of course there are some properties that you should not modify in the restart process such as `simulation/terminate`.

Could you please describe how you are restarting JSBSim ? What is the output you get ? And what is the output you would expect to get instead ?

galleon, 2019-11-18:
So we have define a property Catalog that is created in https://github.com/galleon/gym-jsbsim/blob/master/gym_jsbsim/catalogs/jsbsim_catalog.py
The catalog contains all properties.

A call to ```get_sim_state``` return a dict containing all properties and their values - this is what you mentioned above.

This state is then serialized on our side.

When we want to restart a simulation, we read the previously stored state and call ```set_sim_state```

We then parse the values to identify those that need to become initial conditions (ic). In fact restarting a simulation means starting a new simulation using the last property values as initial conditions.

galleon, 2019-11-24:
In order to better demonstrate what I am doing, I have created this [notebook](https://gist.github.com/galleon/5ba105b699d10b315f5cf4139ce814fa).

When running it the trajectories don't start from my initial location. I do not understand why ?
What is ```ic/lat-geod-deg```compared to ```ic/lat-gc-deg```?

[image]

seanmcleod, 2019-11-24:
> what we want is to store the internal state of JSBSim after a few time step

There is some state though that isn't captured in the property tree. For example, what about the state of any filters? So you can grab the last output value from the filter via the property tree but you won't capture the internal state of the filter in terms of the set of previous values that it stores internally.

bcoconni, 2019-11-27:
> What is `ic/lat-geod-deg` compared to `ic/lat-gc-deg`?

The former sets the initial geodetic latitude in degrees and the latter the initial geocentric latitude in degrees. See on Wikipedia [the difference between geodetic and geocentric latitudes](https://en.wikipedia.org/wiki/Latitude#Geodetic_and_geocentric_latitudes).

You are not supposed to set both of them. In case you do, the last value set will prevail.

bcoconni, 2019-11-27:
In addition, the error is probably not in JSBSim but rather in the code that you use to display the path on the map. In cartography, it is the WGS84 geodetic latitude that is generally used. As a consequence, the latitude saved in `self.path` should be the geodetic latitude i.e. the latitude from the property `position/lat-geod-deg`:
```python
lat = self.exec.get_property_value('position/lat-geod-deg') # Use the WGS84 geodetic latitude
lon = self.exec.get_property_value('position/long-gc-deg')

speed = self.exec.get_property_value('velocities/vc-fps')

self.path.append([lat, lon])
```

bcoconni, 2019-12-14:
@galleon As a gentle reminder, @seanmcleod and myself took the time to answer your questions, it would be very much appreciated if you were polite enough to tell us if that fixes your problem or not. Especially since this is the second time you are behaving as such.

This is the last call before closing this ticket.

bcoconni, 2020-01-03:
Contact lost with issuer : issue closed and locked.

closed 2020-01-03 by bcoconni