JSBSim-Team/jsbsim #195: Preposterous Windmilling Behavior

Octal450, 2019-07-07:
[image]
^ IDG A32X
Hi guys -- even at a high speed, this makes no sense. At low speeds too, the windmilling is too high. This can be see here:
[image]
^ IDG MD-11

I will get the values of what it should be about windmilling at. N2 seems plausible, but N1 is just nonsensical.
Any thoughts?

Kind Regards,
Josh

bcoconni, 2019-07-08:
This may be unrelated but as I mentioned in the issue #178
> `N1` and `N2` are hard coded to be respectively equal to `qbar/10` and `qbar/15` when the engine is off. The mass airflow would more relevant than the dynamic pressure for that matter.

During windmilling it can be assumed that the airflow is proportional to the ram airflow which is proportional to `in.Density` (air density) times the aircraft velocity forward velocity `in.aeroUVW(eU)`. The file `src/models/propulsion/FGTurbine.cpp` could be modified as shown below:

```diff
@@ -190,8 +190,9 @@ double FGTurbine::Off(void)
   FuelFlow_pph = Seek(&FuelFlow_pph, 0, 1000.0, 10000.0);
   // some engines have inlets that close when they are off. So, if a flag is true disable windmilling
   if (disableWindmill == false) {
-    N1 = Seek(&N1, in.qbar/10.0, N1/2.0, N1/N1_spindown);
-    N2 = Seek(&N2, in.qbar/15.0, N2/2.0, N2/N2_spindown);
+    double AirFlow = in.Density * in.AeroUVW(eU);
+    N1 = Seek(&N1, AirFlow/10.0, N1/2.0, N1/N1_spindown);
+    N2 = Seek(&N2, AirFlow/15.0, N2/2.0, N2/N2_spindown);
   } else {
     N1 = Seek(&N1, 0, N1/2.0, N1/N1_spindown);
     N2 = Seek(&N2, 0, N2/2.0, N2/N2_spindown);
```
and the constants `10.0` and `15.0` might need to be updated to reach the expected rotation speeds `N1` and `N2`.

If you can compile FlightGear could you please check the above and let me know if that fixes the problem ?

Octal450, 2019-07-08:
Dear Bertrand, thank you for your response. I do not personally have time for compiling, but I know someone who may be able to test this for me, I will get back to you. Thanks

Kind Regards,
Josh

legoboyvdlp, 2019-07-09:
Didn't I add a flag to disable windmilling?

bcoconni, 2019-07-09:
@legoboyvdlp
My understanding is that we no longer discuss about how to disable
windmilling (as in #178) but rather how to set it up correctly.

legoboyvdlp, 2019-07-09:
Ah, I see, thanks. Ok, sure - I can do that.

Octal450, 2019-07-13:
Bertrand,
It looks like they are now 0.319 and 0.125 - which isn't quite right.
I also believe the N1 factor should be changed from 10 to 20, so that the N1 behaves more properly.

We will test these factors with the qbar code and see if the ratios seem better.

Kind Regards,
Josh

bcoconni, 2019-07-13:
> It looks like they are now 0.319 and 0.125 - which isn't quite right.

I guess you are talking about N1 and N2 ? What are the values you are expecting ? Where are they coming from ?
> I also believe the N1 factor should be changed from 10 to 20, so that the N1 behaves more properly.

As I mentioned above, the coefficients by which the mass air flow is divided likely need some adjustments.
> We will test these factors with the qbar code and see if the ratios seem better.

The dynamic pressure (aka `qbar`) is irrelevant for windmilling. The engine behavior is driven by the air flow as I said earlier: `qbar` is increasing as the square of the velocity while the air flow is increasing linearly with the velocity. If you continue to use the dynamic pressure, `N1` and `N2` will increase too much at higher velocities.

Octal450, 2019-07-13:
Hi Bertrand,
N2 and N1 respectively, according to lego - to me that is odd numbers - without changing the factor, it is strange that they have reversed behavior. I do wonder if it was actually N1 and N2 - I will ask for another test.

Yes, the reason I would test that on qbar is so that I can see the ratio between N1 and N2 to check the coefficient - since it seems after your changes the numbers are very very low and it is not easy to check.

Kind Regards,
Josh

bcoconni, 2019-07-13:
> the reason I would test that on qbar is so that I can see the ratio between N1 and N2 to check the coefficient

Since the value N1 is equal to `x/10` and N2 is equal to `x/15` then whatever is `x` the ratio between N2 and N1 will always be 1.5 whether `x=qbar` or `x=AirFlow`. If you find otherwise then there is most likely an error in the code you are testing.

> it seems after your changes the numbers are very very low and it is not easy to check.

As I already mentioned, the constants 10 and 15 probably need to be tweaked. If you find the value N1 and N2 to be too low then you need to replace the constants 10.0 and 15.0 by smaller values.
```c++
    double AirFlow = in.Density * in.AeroUVW(eU);
    N1 = Seek(&N1, AirFlow/10.0, N1/2.0, N1/N1_spindown);y
    N2 = Seek(&N2, AirFlow/15.0, N2/2.0, N2/N2_spindown);
```
Also I would like to mention that the windmilling engine in `FGTurbine` does not produce drag (or negative thrust) which is much more unrealistic than the fact that the values of N1 and N2 are erroneous.

Octal450, 2019-07-13:
I see what you mean, sorry - I had a stupid moment there - of course the coefficient will change with a new value being used not qbar.
So I'll get back to you on the numbers.

Kind Regards,
Josh

bcoconni, 2019-08-15:
@Octal450 
> So I'll get back to you on the numbers.

Any progress on this topic ?

Octal450, 2019-08-15:
Apologies, I've been particular busy with my life especially my health. I'll try to get it as soon as possible for you 

Kind Regards,
Josh

bcoconni, 2019-08-15:
> Apologies, I've been particular busy with my life especially my health. I'll try to get it as soon as possible for you

Oh ! Sorry, my apologies. There is no rush answering this topic.

Octal450, 2019-08-15:
No worries! I am glad you did as I had forgotten in the time. I've written it in my to-do.

Kind Regards,
Josh

Octal450, 2019-09-16:
Apologies for further delays - my health is recovered and I am now back at FG so I'll definitely get this soon.
Josh

mike402, 2019-12-14:
By the way, I don't know if this is related, but...

If I throw the model up in the air or disable `seized` property in flight, engine will not windmill unless it has either had `starter=1` or been running.

bcoconni, 2021-09-17:
@Octal450, speaking of windmilling in issue #494. Is there any progress on this issue ?

Octal450, 2021-09-17:
Hi,
Sorry for slowly. With my health everything been all over the place.

I need to do more research first before I can make a recommendation. I got some conflicting info but didn't have time to check further.

Thanks for your patience. 
Josh

seanmcleod, 2021-10-09:
Here is a potentially useful paper (I haven't paid the $25 to download it) - [Modelling and Analysis of Turbofan Engines Under Windmilling Conditions](https://arc.aiaa.org/doi/10.2514/1.B34729)

I'm not suggesting that we implement a full high-fidelity windmilling model, but rather that we should be able to find some first or 2nd order approximation from the paper that we can use to generate at least semi-realistic N1/N2 values.

seanmcleod, 2021-11-01:
@bcoconni please take a look at the notebook above when you get a chance.

bcoconni, 2021-11-02:
@seanmcleod looks good to me.

I guess the reason why the curves do not match is that the engine components downstream of the inlet have an influence on the air mass flow and that we cannot simply assume that the air velocity at the inlet equals the aircraft velocity.