JSBSim-Team/jsbsim #196: Turbine fuel consumption

mike402, 2019-07-08:
Hi,

I found the following problems with turbine fuel flow.

1. `fuel-flow-rate-pps` considerably lags behind the change of `thrust-lbs` -- it takes almost half a minute to settle after thrust has. But as far as I've heard, thrust and fuel flow are proportional in the same regime, and flight engineers used flow gauges to match thrust of individual engines in flight?

2. There is no way to make variable TSFC, as `<tsfc>` tag is just a number and is not exposed anywhere in property tree -- so if you have a table of TSFC per flight regime, there is no way to put it to use, and you will have the wrong range for anything but cruise regime.

Kind regards,
Mike

bcoconni, 2019-07-09:
> `fuel-flow-rate-pps` considerably lags behind the change of thrust-lbs -- it takes almost half a minute to settle after thrust has. But as far as I've heard, thrust and fuel flow are roughly proportional in each given regime, and flight engineers used flow gauges to match thrust of individual engines in flight?

`FGTurbine` uses millions of magic numbers and the latency for the fuel flow is determined by some of them.
https://github.com/JSBSim-Team/jsbsim/blob/c0b8e62113492570f211f6510fd075eb65396319/src/models/propulsion/FGTurbine.cpp#L190
https://github.com/JSBSim-Team/jsbsim/blob/c0b8e62113492570f211f6510fd075eb65396319/src/models/propulsion/FGTurbine.cpp#L239
https://github.com/JSBSim-Team/jsbsim/blob/c0b8e62113492570f211f6510fd075eb65396319/src/models/propulsion/FGTurbine.cpp#L253
https://github.com/JSBSim-Team/jsbsim/blob/c0b8e62113492570f211f6510fd075eb65396319/src/models/propulsion/FGTurbine.cpp#L262
These numbers need to be tweaked to obtain the result you are seeking.
> There is no way to make variable TSFC, as `<tsfc>` tag is just a number and is not exposed anywhere in property tree -- so if you have a table of TSFC per flight regime, there is no way to put it to use, and you will have wrong range for anything but cruise.

It is feasible to make the TSFC available as a property however `FGTurbine` uses a corrected version of TSFC that depends on the ambient temperature, `N2` and some more magic numbers.
https://github.com/JSBSim-Team/jsbsim/blob/c0b8e62113492570f211f6510fd075eb65396319/src/models/propulsion/FGTurbine.cpp#L238
So giving a simple access to the TSFC value might not allow to drive the fuel consumption unless you compensate for the formula above.

mike402, 2019-08-05:
But wait, aren't turbines using RPM adjusted to ambient temperature internally? Because I see absolutely no change in n1 and n2 versus flight regime and have to calculate mechanical RPM by myself.

bcoconni, 2019-08-06:
> But wait, aren't turbines using RPM adjusted to ambient temperature internally?

`FGTurbine` **is not** by any means an engineering model that simulates the thermodynamics of a turbine engine. It just guesses numbers that look similar to the parameters of a real engine if you don't look too closely.

You are correct in the fact that `N2` should be corrected by temperature but it should be corrected with the HP compressor inlet temperature not the ambient temperature. Then comes the next problem: how do you assess the air temperature at the inlet of the HP compressor ?
> Because I see absolutely no change in n1 and n2 versus flight regime and have to calculate mechanical RPM by myself.

`N1`and `N2`are driven by a linear law that simply depends on the throttle position. So the law N1=f(N2) is invariably the same linear function no matter what the flying conditions are. As said above, it is not an engineering model.
https://github.com/JSBSim-Team/jsbsim/blob/4c0c0930b3871aeebbc29a24063d85155a7c83d8/src/models/propulsion/FGTurbine.cpp#L143-L144

mike402, 2019-08-14:
Indeed, it makes sense to use "adjusted" RPM in the model and let user calculate the "real" thing from what they have. But I'm having a problem with using user-supplied functions for RPM in FlightGear:

https://github.com/JSBSim-Team/jsbsim/issues/210

As for consumption -- I think one may abuse the tanks' external-flow-pps  but I haven't tried yet:

* `<tsfc>0.0</tsfc>`
* Make a `<fcs_function>` that would calc fuel flow from engine's thrust

mike402, 2019-08-15:
But if I set tsfc to 0 in engine file, I still see fuel-flow-rate-pps of around 0.252 when engine is running.

bcoconni, 2019-08-15:
> But if I set tsfc to 0 in engine file, I still see fuel-flow-rate-pps of around 0.252 when engine is running.

Yeah, this is most likely due to the code below which forces the fuel flow to be at least equal to the idling fuel flow:
https://github.com/JSBSim-Team/jsbsim/blob/4c0c0930b3871aeebbc29a24063d85155a7c83d8/src/models/propulsion/FGTurbine.cpp#L240
and, of course, the idling fuel flow `IdleFF` is hard coded :sob:
https://github.com/JSBSim-Team/jsbsim/blob/4c0c0930b3871aeebbc29a24063d85155a7c83d8/src/models/propulsion/FGTurbine.cpp#L526
I would like to take this opportunity to mention that the SFC is not a monotonically increasing function of N2. In most cases, lower SFCs are obtained at cruise which is the design point of the engines. SFC at idle is therefore higher than at cruise due lower efficiencies of the engine and as a consequence the line of code below is inaccurate
https://github.com/JSBSim-Team/jsbsim/blob/4c0c0930b3871aeebbc29a24063d85155a7c83d8/src/models/propulsion/FGTurbine.cpp#L238

mike402, 2019-08-15:
It seems I can still get what I want by leaving `<tsfc>` alone, but subtracting the default fuel flow from the additional fuel flow I calculate from the TSFC function.

bcoconni, 2019-08-15:
> It seems I can still get what I want by leaving <tsfc> alone, but subtracting the default fuel flow from the additional fuel flow I calculate from the TSFC function.

Excellent ! Can we close this issue then ?
> Maybe TSFC also does not reflect idle power, hence the idleFF?

No, since `IdleFF` is only used to constrain the fuel flow to a minimum value:
https://github.com/JSBSim-Team/jsbsim/blob/4c0c0930b3871aeebbc29a24063d85155a7c83d8/src/models/propulsion/FGTurbine.cpp#L238-L240
It is impossible to get a 'U' shaped curve with this code.

mike402, 2019-08-15:
I guess we can close this, yes, thanks for the info.

closed 2019-08-15 by bcoconni