JSBSim-Team/jsbsim #197: cmake build problem

HanbumKo, 2019-07-12:
Hello, I tried to install, but it showed error.

At first, I followed install instruction in README.md. I download the file
"python3-JSBSim_1.0.0.dev1-495.trusty.amd64.deb"
And there is no response when I click install button.

After trying that, I followed install instruction in JSBSim Reference manual.
But it fails also when I build.

Followings are my history.
git clone https://github.com/JSBSim-Team/jsbsim.git jsbsim-code
cd jsbsim-code
mkdir build
cd build
cmake ..

and shows errors
-- The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 5.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found PythonInterp: /usr/bin/python (found version "2.7.12") 
-- Could NOT find Cython (missing:  CYTHON_EXECUTABLE) 
CMake Warning at CMakeLists.txt:60 (message):
  JSBSim Python module and test suite will not be built

-- Could NOT find CxxTest (missing:  CXXTEST_INCLUDE_DIR CXXTEST_PYTHON_TESTGEN_EXECUTABLE) 
-- Could NOT find Doxygen (missing:  DOXYGEN_EXECUTABLE) 
-- Found LSB_Release: /usr/bin/lsb_release
-- Configuring done
-- Generating done
-- Build files have been written to: /home/episci/Downloads/jsbsim-code/build

How can I deal with this?

bcoconni, 2019-07-13:
> How can I deal with this?

There is nothing wrong with the messages about Cython, CxxTest and Doxygen. If you proceed with `make`, the C++ code should build and you should obtain an executable named `JSBSim` in `/home/episci/Downloads/jsbsim-code/build/src`.

Now if you plan to use the Python interface, you should install Cython prior to the build. The 2 other packages (CxxTest and Doxygen) are only needed if you plan to run the C++ unit tests and/or build the documentation that is available at https://jsbsim-team.github.io/jsbsim/

HanbumKo, 2019-07-15:
Oh, I misunderstood 'JSBSim Python module and test suite will not be built' to that build will be failed. Let me try. Thanks!!

closed 2019-07-15 by HanbumKo