JSBSim-Team/jsbsim #199: Python documentation?

HanbumKo, 2019-07-22:
I installed jsbsim and read user manual. And what I understood is that we can control flight through .xml file or with C++ code.
But I've wanted to use Python code not C++ to collaborate with Pytorch or Tensorflow. Is there any python document here?

bcoconni, 2019-07-23:
> Is there any python document here?

Unfortunately not much. However, the simplest Python code to run a script is
```python
import jsbsim

fdm = jsbsim.FGFDMExec('.', None)
fdm.load_script('scripts/c1721.xml')
fdm.run_ic()

while fdm.run():
  pass
```
Alternatively you can use the file https://github.com/JSBSim-Team/jsbsim/blob/master/python/JSBSim.py as a more elaborate example.

HanbumKo, 2019-07-24:
Thanks,
So if I want to change flight parameters dynamically using python, I have to load .xml file and fix the .xml file manually? But I'm not sure it can be changed dynamically(I mean, in run time).

bcoconni, 2019-07-24:
> But I'm not sure it can be changed dynamically(I mean, in run time).

Properties can be read and modified dynamically :
```python
  # Read the latitude
  latitude = fdm['position/lat-gc-rad']
  # Set the gravity model
  fdm['simulation/gravity-model'] = 0
```

HanbumKo, 2019-07-25:
That's exactly what I wanted to know!! Thanks

HanbumKo, 2019-08-05:
OK, now I checked the code works well, but I want to interface JSBSim with FlightGear in real time.
Using 
`JSBSim scripts/Short_S23_2.xml data_output/flightgear.xml --realtime --nice`
works well in real time.
How can I run like this using python?

bcoconni, 2019-08-06:
A script can be loaded in an FDM instance by the `load_script`method and the output file by the method `set_output_directive`:
```python
  fdm.load_script('scripts/ShortS23_2.xml')
  fdm.set_output_directive('data_output/flightgear.xml')
```
This should executed before running the initial conditions i.e. before `fdm.run_ic()` is executed.

The flag `--realtime` indicates that the script should be executed in real time. For example, if your script is running at a frequency of 120 Hz, the next time step should be executed after one 120th of a second has elapsed since the previous time step. JSBSim does not manage that so you have to write the Python code that simulates that yourself.

The flag `--nice` indicates that JSBSim should use as few CPU power as possible while waiting for the next time step to be executed. This can be emulated by repeatedly calling the `sleep` system function between each time step.  JSBSim does not manage that either so you have to write the Python code to emulate that as well.

HanbumKo, 2019-08-21:
Thanks a lot... Now I know what this system is. But I have a lot of questions. But now it's OK. Let me close this issue and I would post new issue when I want to know other things. Thanks!!

closed 2019-08-21 by HanbumKo

bcoconni, 2019-09-01:
@HanbumKo 
FYI the Python module API is now documented and can be accessed from https://jsbsim-team.github.io/jsbsim/python/index.html

viren3999, 2021-01-16:
> FYI the Python module API is now documented and can be accessed from https://jsbsim-team.github.io/jsbsim/python/index.html

@bcoconni Sir, this page seems to be **blank** when clicked on particular classes. Has it been moved to some other page or website ?

bcoconni, 2021-01-16:
> Sir, this page seems to be blank when clicked on particular classes. Has it been moved to some other page or website ?

@viren3999 No, it's a bug in the Python docs generator. I'll investigate.
Could you please report that in a new separate issue ? This issue was a different topic and is now closed.
Thanks.