JSBSim-Team/jsbsim #200: Actuator crash - hardover

legoboyvdlp, 2019-07-22:
Hi,
If I set a hardover fail and no minimum value is set, FlightGear crashes. Perhaps a nil check is needed on line 159 of FGActuator?

  if (fail_hardover) Input = Input < 0.0 ? ClipMin->GetValue() : ClipMax->GetValue();

bcoconni, 2019-07-24:
I pushed a tentative fix to FlightGear. Let me know if that fixes the problem.

legoboyvdlp, 2019-07-25:
Hi,
Thank you!

legoboyvdlp, 2019-07-27:
I have compiled last night and hope to try it today. Sorry for the delay.

legoboyvdlp, 2019-07-27:
Unfortunately not, it still crashes.

bcoconni, 2019-07-30:
OK I'll investigate further.

bcoconni, 2019-08-03:
> Unfortunately not, it still crashes.

That's strange, according to the regression test added by the commit e743b3b, the bug is now fixed. Are you sure you have tested the last version of FlightGear ?

legoboyvdlp, 2019-08-04:
Hmm - I think so, but perhaps my compiler was misbehaving. I will try it again.

bcoconni, 2019-08-04:
> Also, surely a hardover means not the neutral position but the minimum or maximum position is reached and the actuator sticks there? Did I misunderstand the test?

No but for that to happen you need to specify the minimum and maximum position i.e. you need to define a `<clipto>` element such as
https://github.com/JSBSim-Team/jsbsim/blob/de6bbb1fa0785142dafe5eb3825967d28bd641d7/aircraft/c172x/c172x.xml#L355-L366
If the element `<clipto>` is missing then the actuator goes to its neutral position for the lack of  better option.

mike402, 2019-08-05:
I thought actuators, integrators etc didn't respect `<clipto>` internally and just silently kept going beyond the value?

bcoconni, 2019-08-06:
> I thought actuators, integrators etc didn't respect `<clipto>` internally and just silently kept going beyond the value?

**????**
Where this assumption is coming from ? Have you experienced some cases where `<clipto>` was ignored ? In that case this should be reported as a bug.

mike402, 2019-08-07:
~~~
 <property>asdf/inp</property>
 <channel name="Test">

  <integrator name="asdf/int_clip">
   <output>asdf/int_clip_</output>
   <input>asdf/inp</input>
   <c1>1.0</c1>
   <clipto>
    <min>-1.0</min>
    <max>1.0</max>
   </clipto>
  </integrator>

  <integrator name="asdf/int">
   <output>asdf/int_</output>
   <input>asdf/inp</input>
   <c1>1.0</c1>
  </integrator>

  <actuator name="asdf/act_clip">
   <output>asdf/act_clip_</output>
   <input>asdf/inp</input>
   <rate_limit>1.0</rate_limit>
   <clipto>
    <min>-1.0</min>
    <max>1.0</max>
   </clipto>
  </actuator>

  <actuator name="asdf/act">
   <output>asdf/act_</output>
   <input>asdf/inp</input>
   <rate_limit>1.0</rate_limit>
  </actuator>

 </channel>
~~~

1. Set asdf/inp to 10 and wait for some time, the unclipped integrator and actuator will go beyond the <clipto> values.
2. Set asdf/inp to -10.

Expectation: clipped integrator and actuator would immediately start going back from their limit.

Reality: they stay in sync with the unclipped versions, indicating that clipto is not respected by internal logic and just clips the output.

bcoconni, 2019-08-10:
@mike402 
I'll investigate that. Meanwhile would you mind opening a new issue about that topic and delete your post above to avoid polluting the topic from @legoboyvdlp ?

Thanks.

mike402, 2019-08-10:
No problem, I moved the post.

bcoconni, 2019-08-15:
@legoboyvdlp, have you come to a conclusion regarding to this issue ?

legoboyvdlp, 2019-08-19:
No, sorry.

bcoconni, 2019-10-31:
@legoboyvdlp 
Have you had the opportunity to check if that issue is fixed ?

legoboyvdlp, 2019-11-01:
Yeah it goes to value 0 now when I set hardover.

Sorry, this was long overdue, my apologies.

closed 2019-11-16 by bcoconni