JSBSim-Team/jsbsim #201: Python bindings segfault when deallocating an instance of FGFDMExec when multiple are instantiated in memory

venabled, 2019-07-23:
FGFDMExec must depend on something that has only a single instance in memory. When instantiating two exec classes, deallocation of the first works, then the second segfaults. 

Example output: 

```
In [1]: import jsbsim                                                                                                                                                                                                                                    

In [2]: mgr = jsbsim.FGPropertyManager(new_instance=True)                                                                                                                                                                                                      

In [3]: fdm = jsbsim.FGFDMExec('/home/venabled/source/jsbsim', None)                                                                                                                                                                                     

     JSBSim Flight Dynamics Model v1.0.0rc1 Jul 23 2019 12:40:46
            [JSBSim-ML v2.0]

JSBSim startup beginning ...

In [4]: fdm2 = jsbsim.FGFDMExec('/home/venabled/source/jsbsim',mgr)                                                                                                                                                                                     

     JSBSim Flight Dynamics Model v1.0.0rc1 Jul 23 2019 12:40:46
            [JSBSim-ML v2.0]

JSBSim startup beginning ...

In [5]: fdm = []                                                                                                                                                                                                                                         

In [6]: fdm2 = []                                                                                                                                                                                                                                        
Segmentation fault (core dumped)
```

Validated on python3.6 and 3.7 with master, and on the 1.0rc, both built from source and using the module provided in releases. Validated on arch-linux (amd-64), (Anaconda python 3.6, 3.7, and system Python 3.7), and macOS Mojave (Anaconda Python 3.6 and 3.7)

bcoconni, 2019-07-23:
Yes, this is a known issue that is due to the usage of a `static` pointer in `FGLocation`
https://github.com/JSBSim-Team/jsbsim/blob/0ec1cc873ece3fb8d2894794128c692d46290687/src/math/FGLocation.h#L611-L613
For now, this can be ignored unless this is an obstacle to your application.
I will fix this issue later.

venabled, 2019-07-24:
Copy, will work around for now using multiprocessing. If I end up needing it for the application I can help with the fix.

crichardson332, 2019-10-10:
I'm currently working with a multi-vehicle simulation engine that utilizes multi-threading to efficiently run simulations of thousand of flying vehicles for robotics applications. The multi-threading is important especially when running large batch simulations while doing machine learning. We're using JSBSim as our vehicle simulation back end.

This bug is a huge problem for us, since it causes segfaults whenever we have vehicles that are destructed (which happens often during our game simulations, when vehicles are destroyed for various reasons and their resources are destructed). Currently we have to maintain a patch for JSBSim that removes this bug, but it's annoying to have to keep it around. Is there any plan to fix this in the near future? We want to get back onto mainline master for JSBSim but we can't do that until this bug is fixed. Thanks.

opt12, 2019-11-29:
I also stumbled into the "no multithreading" issue recently, when setting up a machine learning environment using JSBSim based on https://github.com/Gor-Ren/gym-jsbsim 

At least, I was warned, that mutlithreading won't work for the moment.

It would be a huge win for machine learning applications to have the possibility to instantiate more than one instance of `FGFDMExec` A lot of modern algorithms rely on instantiating multiple instances of the simulation engine. Be it for diversity in the learning samples during training or be it for eindependent evaluation while the learning is still running.

@crichardson332 , @venabled : Looks like you have a workaround/patch for this multithreading issue for JSBSim. Can you please make this available with some words of explanation. this would be a big help for me.

Regards, Felix

bcoconni, 2019-12-01:
Thanks all for the report.
FYI I am currently working on this issue. I'm trying to avoid the quick & dirty fix option, so it involves some code refactoring which might take some time before getting a fix.

bcoconni, 2019-12-01:
@venabled , @crichardson332 and @opt12 
I have pushed commits (mainly 2661d7c) that have removed the static pointers. You should no longer experience segfaults when using several FDM instances simultaneously.

The Python wheel packages are available in the [2019 rolling release section](https://github.com/JSBSim-Team/jsbsim/releases/tag/Rolling-release-v2019) for Windows 64bits and Linux 64bits. They may be a bit rough around the edges at the moment so let me know if you encounter any problems.

opt12, 2019-12-03:
@bcoconni Big Kudos to Bertrand! Works for me.

agodemar, 2019-12-06:
@venabled @crichardson332 @opt12 
I would love to learn more about how you are using JSBSim in your projects, if possible. 
Please feel free to PM me.

bcoconni, 2019-12-14:
Well, I guess this ticket is closed now.
Thanks you all for the report and the testing.

closed 2019-12-14 by bcoconni

crichardson332, 2019-12-20:
I haven't had a chance to try the fix yet and won't be able to for a few weeks - we do a lot of flight testing and can't update to the newest commit for a bit. But if we encounter any issues when we do, I will make tickets for it. Thanks!