JSBSim-Team/jsbsim #202: How to use Xcode Mac os

HanbumKo, 2019-07-27:
I gave up using python binding because there is no documentation, and decided to use C++.
I build with CMake, but then I have no idea how to start with Xcode.

Is there anyone who are using Xcode for C++?

bcoconni, 2019-07-30:
> I gave up using python binding because there is no documentation, 

If you could describe how you would like to interact with JSBSim, perhaps we could provide you with some help ?
> Is there anyone who are using Xcode for C++?

Sorry, I'm developing on Linux so I cannot provide support for that matter.

HanbumKo, 2019-08-05:
When I want to use python, I did "python setup.py install" or "pip install jmavsim". And I check "import jmavsim" works well in python code.

But how can I use when I want to C++? I've only used python packages at other projects. I have no idea to deal with C++.

I want to use like the example
```
#include <FGFDMExec.h> // << error

int main(int argc, char **argv) 
{ 
  JSBSim::FGFDMExec FDMExec; 
  bool result = true; 
  FDMExec.LoadScript(argv[1]); 
  while (result) result = FDMExec.Run();
}
```

I builded using cmake. But how can I use after "make install"?

Sorry for my confusing explanation.

bcoconni, 2019-08-06:
I guess this topic is now on hold until the issue #199 is resolved ?

HanbumKo, 2019-08-21:
Yes. Python issue is solved, so I don't need to do this.

closed 2019-08-21 by HanbumKo