JSBSim-Team/jsbsim #203: Catapult launch with a pitch angle

alexander-kuptsov, 2019-07-31:
Currently I am developing sim for a small UAV which has a catapult as well as a bungee launch capabilities.

Working on that I faced with the following issue and googling didn't help much.

1. The catapult has a positive pitch (10-15 deg), so how to setup FRAME or may be initial conditions to handle that? I do not have a bogey for example and AFAIK JSBSim expects flat surface for launch.

Thanks for any assistance!

bcoconni, 2019-08-04:
Hi,

Unless you want ot simulate the catapult course, you can start the simulation at the very moment at which the UAV is released from the catapult by defining an appropriate velocity vector in the initial conditions.
See for example the initial conditions of the canon ball script:
https://github.com/JSBSim-Team/jsbsim/blob/de6bbb1fa0785142dafe5eb3825967d28bd641d7/aircraft/ball/reset01.xml#L1-L18
Here the ball has a velocity of 500 ft/s with an angle of 40 degrees with respect to the horizontal plane (defined by the tag `<theta>` which is the pitch angle) and a heading of 40 degrees (tag `<psi>`)

alexander-kuptsov, 2019-08-04:
Hi, 
Thank you for the advice, sounds quite feasible, but at the moment UAV uses ArduPilot JSBSIM SITL and all interactions go throught ArduPilot simulation code. That couses some limitations and lack of flexibility to run simulation at the particular moment, moreover if we take in account network delays the task would not look like obvious thing))

So, is there any option to launch UAV with given direction (angle), and force (impulse)?

Thank you in advance!

PS:  The second issue I have is flying on a negative relative altitude - for example if we launch an UAV on a hill then my relative altitude can be negative. Is there any option to manage that?

bcoconni, 2019-08-15:
> is there any option to launch UAV with given direction (angle), and force (impulse)?

If you have access to JSBSim properties you can set the initial conditions using the following properties:
- `ic/u-fps` to 500 (velocity in the body frame along the X axis)
- `ic/theta-deg` to 40.0 (pitch angle)
- `ic/psi-true-deg` to 90.0 (true heading)
- `ic/h-sl-ft` to 10000 (altitude above Sea Level)

Then you can trigger a reset by setting `simulation/reset` to 1. The aircraft parameters will then be reset to the initial conditions that were previously set.

Alternatively you can define an impulse force with an external force such as in the weather balloon model:
https://github.com/JSBSim-Team/jsbsim/blob/4c0c0930b3871aeebbc29a24063d85155a7c83d8/aircraft/weather-balloon/weather-balloon.xml#L320-L345
However for that option as well you will need an access to the properties to manage the external force/impulse

alexander-kuptsov, 2019-08-15:
Thanks a million! Will try this approach
Thank you so much!

bcoconni, 2019-08-21:
> Thanks a million! Will try this approach
Thank you so much!

You're welcome 😃 Shall we consider this issue closed ?

alexander-kuptsov, 2019-08-22:
I think yes at the moment) thanks again for your help!

bcoconni, 2019-08-22:
Do not hesitate to open a new issue if you need further help.

closed 2019-08-22 by bcoconni