JSBSim-Team/jsbsim #204: <table>   IndependentVar property, ${PROPERTY} in Table definition is not defined.

mike402, 2019-08-04:
If you define a table like 
`
  <fcs_function name="asdf">
   <function>
    <table>
     <independentVar lookup="row">hjkl</independentVar>
     <tableData>
0.0 0.0
1.0 2.0
     </tableData>
    </table>
   </function>
  </fcs_function>
`
and if `hjkl` is not a `<property>` but a filter like
`
<property>hjkl_input</property>
  <pure_gain name="hjkl">
   <input>hjkl_input</input>
   <gain>1.0</gain>
  </pure_gain> 
`
FlightGear will die with an error like:

`
In file /oops.xml: line 123

  IndependentVar property, hjkl in Table definition is not defined.
`

And if you define `hjkl` as `<property>`, the problem will go away, but you will get a warning about double definition as discussed in https://github.com/JSBSim-Team/jsbsim/issues/140

Other functions don't require `<property>` to be used and work with `name=""` filters directly, so I believe this is a bug.

bcoconni, 2019-08-06:
I cannot reproduce this issue with the code from JSBSim `master` branch. However the code in `master` is now different from the JSBSim code in FlightGear in response to the requests you made in issue #140 .

I will propagate the bleeding edge code to FlightGear later this week and you will be able to tell me if the issue you've just reported is still applicable.

mike402, 2019-08-07:
Thanks, I'll try!

bcoconni, 2019-08-10:
I just pushed the up to date code to FlightGear (except the last 2 commits 4c0c093 and 5776551 which are WIP).
Could you please test and let me know the status for this issue ?
Thanks.

mike402, 2019-08-14:
Tried latest synced JSBSim, and I still get the errors.

bcoconni, 2019-08-15:
> tried latest synced JSBSim, and I still get the errors.

OK
The good news: I could reproduce the issue as you described it.

The bad news: it is not a bug but a subtlety in JSBSim. When a property is declared with `name=""`, JSBSim prepends it with `fcs/` (unless there is already a `/` in the name). So your code can be fixed if you change the property name in the `<independentVar>` element by `fcs/hjkl`
```xml
<fcs_function name="asdf">
  <function>
  <table>
    <independentVar lookup="row">fcs/hjkl</independentVar>
    <tableData>
      0.0 0.0
      1.0 2.0
    </tableData>
  </table>
  </function>
</fcs_function>
```
The other option (but I know you dislike it :wink:) is to use `<output>` instead of `name=""` for the `<pure_gain>` element:
```xml
<pure_gain>
  <input> hjkl_input </input>
  <gain> 1.0 </gain>
  <output> hjkl </output>
</pure_gain>
```
In that case, JSBSim does not prepend `hjkl` with `fcs/` and you can refer later refer to this property as `hjkl`.

mike402, 2019-08-15:
But if that was the case, a lot of things would break because people
(including me) use `name="path/to/property"` and intend it to end up
in `/fdm/jsbsim/path/to/property` -- as in, `name=""` without `/` is
path relative to the root of jsbsim tree -- and not
`/fdm/jsbsim/fcs/path/to/property`. I think you mean `/fdm/jsbsim`,
because I don't see anything of my stuff under `/fdm/jsbsim/fcs/`,
except what I explicitly put there with `name="fcs/blabla"`.

So is name="" supposed to be human-readable and unique name of the
filter that will be a bit mangled and put under `fcs/`, and one should
use `<output>` instead?

I, personally, have no strong preference of `name=""` vs `<output>`
and would use whichever is "right" and guaranteed to not suddenly
change in the future -- but I wouldn't like to give each filter a
human-readable `name=""` in addition to property path, would double
the already high verbosity.

bcoconni, 2019-08-15:
> But if that was the case, a lot of things would break because people (including me) use `name="path/to/property"` and intend it to end up in `/fdm/jsbsim/path/to/property` -- as in, `name=""` without `/` is path relative to the root of jsbsim tree -- and not `/fdm/jsbsim/fcs/path/to/property`.

You did not read carefully what I wrote. I explicitly stated that when the property name includes a slash `/` then it **is not** mangled with `fcs/`. The property name in your example contains slashes (`path/to/property`) so it is not prepended with `fcs/`

The code at stake is just below, you can check by yourself
https://github.com/JSBSim-Team/jsbsim/blob/4c0c0930b3871aeebbc29a24063d85155a7c83d8/src/models/flight_control/FGFCSComponent.cpp#L249-L257
> So is `name=""` supposed to be human-readable and unique name of the filter that will be a bit mangled and put under `fcs/`, and one should use `<output>` instead?

Did I say that `<output>` should be used ? I offered you 2 options to fix the problem you have reported. Which option you use is your choice.

> I, personally, have no strong preference of `name=""` vs `<output>` and would use whichever is "right" and guaranteed to not suddenly change in the future

The feature we are discussing has been introduced by the commit 06c1dec117 which dates back from May 2, 2006 i.e. more than 13 years ago. It has not "suddenly changed" as you seem to imply. And before that commit was introduced, property names in `name=""` were **always** prepended with `fcs/` whether they contained slashes or not. By the way I encourage you to read the commit message of 06c1dec117 as it explains quite well the motivation behind the current situation.

Finally, have you checked if the options I submitted previously actually fix your problem or not ?

mike402, 2019-08-15:
Sorry, I had misread it like "it always prepends `fcs/`" and got very scared. O_O

Both suggested fixes don't fix the error.

But in my example I did not intend to put things in `fcs/`, I did it by accident. It
complains about missing property even if paths do contain slashes and end up in intended places, here's updated example:

~~~
 <property>test/input</property>

 <channel name="Test">

  <fcs_function name="test/asdf">
   <function>
    <table>
     <independentVar lookup="row">test/hjkl</independentVar>
     <tableData>
0.0 0.0
1.0 2.0
     </tableData>
    </table>
   </function>
  </fcs_function>

  <pure_gain name="test/hjkl">
   <input>test/input</input>
   <gain>1.0</gain>
  </pure_gain>

 </channel>
~~~

bcoconni, 2019-08-15:
I can reproduce the issue and for that particular instance, order matters because if you define the `<pure_gain>` element before the `<fcs_function>` then the error message vanishes.

Tested code that works:
```xml
  <channel name="Test">
    <pure_gain name="test/hjkl">
      <input>test/input</input>
      <gain>1.0</gain>
    </pure_gain>

    <fcs_function name="test/asdf">
      <function>
        <table>
          <independentVar lookup="row">test/hjkl</independentVar>
          <tableData>
            0.0 0.0
            1.0 2.0
          </tableData>
        </table>
      </function>
    </fcs_function>
  </channel>
```
I guess that is because the property `test/hjkl` is defined when JSBSim parses the `<pure_gain>`. If JSBSim parses the `<fcs_function>` first then the property is not yet defined and an error message is issued.

mike402, 2019-08-15:
Ah, that explains it! But because other filters don't require being after the other filter, this is counterintuitive.

bcoconni, 2019-08-16:
> other filters don't require being after the filters used as input,

I have tested the following FCS components and **they all** issue the error message `IndependentVar property, test/hjkl in Table definition is not defined.` when they are defined **after** the `<table>` component:
- `<pure_gain>`
- `<summer>`
- `<distributor>`
- `<waypoint_heading>`
- `<waypoint_distance>`
- `<actuator>`
- `<linear_actuator>`
- `<lag_filter>`
- `<pid>`
- `<kinematic>`
- `<switch>`
- `<deadband>`
- `<angle>`
- `<accelerometer>`
- `<gyro>`
- `<magnetometer>`

So my tests do not support your statement. Could you provide me with an example where the error message is not issued **and** the filter/component is defined after the `<table>` component ?

andgi, 2019-08-16:
I think the issue is with the implementation of <table> and not the other components. That is, it is one point where the modification to support late appearing properties (i.e. checked at the first run time step rather than during construction) I did some years back have not been inserted.

mike402, 2019-08-16:
Well I have dozens of systems and hundreds of filters in my aircraft, I would not have managed to avoid the problem by luck.

But one could try that by reversing the order of `n2-mechanical-theoretical`, `n2-mechanical-limited` and `n2-mechanical` in https://gitlab.com/mdanil/Tu-144/blob/0.10.2/Tu-144-jsbsim/propulsion/RD-36_0.xml#L279, and the model would still run in FG.

andgi, 2019-08-16:
To be specific your message is probably generated by line 188-191 in the constructor of FGTable.cpp:
`if (node == 0) {
    cerr << axisElement->ReadFrom();
    throw("IndependentVar property, " + property_string + " in Table definition is not defined.");
}`
where node is of type FGPropertyNode*. So if that property doesn't exist when the &lt;table&gt; is constructed it will fail hard. That's why that property must be created before, either by a preceding <property> element or a preceding component.

bcoconni, 2019-08-17:
@andgi 
> I think the issue is with the implementation of `<table>` and not the other components. That is, it is one point where the modification to support late appearing properties (i.e. checked at the first run time step rather than during construction)

I just pushed the commit 63045697 that will allow late binding of `<table>` independent vars as you suggested. According to my tests, it successfully removes the requirement to have the property defined before the `<table>` definition is parsed.

bcoconni, 2019-08-17:
@mike402 
> I have tried reversing the order of `n2-mechanical-theoretical`, `n2-mechanical-limited` and `n2-mechanical` which take each other as input in https://gitlab.com/mdanil/Tu-144/blob/0.10.2/Tu-144-jsbsim/propulsion/RD-36_0.xml#L279, and the model would still run in FG.

For the record, `n2-mechanical-theoretical` is not used in your model as a `<table>` independent var (nor is `n2-mechanical-limited`). And the remaining property `n2-mechanical` is defined as a property at line 21 https://gitlab.com/mdanil/Tu-144/blob/0.10.2/Tu-144-jsbsim/propulsion/RD-36_0.xml#L21. Therefore it is not surprising that you did not meet the problem we are discussing since none of the properties you mentioned meet the conditions to trigger the error message.

mike402, 2019-08-17:
> For the record, `n2-mechanical-theoretical` is not used in your model as a `<table>` independent var (nor is `n2-mechanical-limited`). And the remaining property `n2-mechanical` is defined as a [...]

Sorry I had misread you again! >.< I thought you were saying that other components were issuing the same error as `<table>` for you -- I was trying to show that the error indeed only happens with `<table>`.

bcoconni, 2019-08-17:
@mike402 
> Sorry I had misread you again! >.<

No problem. FYI I just pushed the commit to FlightGear `next` branch for testing purposes. Let me know if that helps.

mike402, 2019-08-18:
It works, thank you for the fix!

bcoconni, 2019-08-21:
> It works, thank you for the fix!

Excellent ! Issue closed.

closed 2019-08-21 by bcoconni