JSBSim-Team/jsbsim #206: <actuator>, <integrator>, possibly other <*filter>: no respect for <clipto>

mike402, 2019-08-10:
Hi,

I noticed this a long ago, but didn't think it was unintentional. Many filters which involve integration are not internally stopped at `<clipto>` limits, instead they keep going beyond those values inside, and only the visible result is clipped. IMO this is very developer-error-prone.

~~~
 <property>asdf/inp</property>
 <channel name="Test">

  <integrator name="asdf/int_clip">
   <output>asdf/int_clip_</output>
   <input>asdf/inp</input>
   <c1>1.0</c1>
   <clipto>
    <min>-1.0</min>
    <max>1.0</max>
   </clipto>
  </integrator>

  <integrator name="asdf/int">
   <output>asdf/int_</output>
   <input>asdf/inp</input>
   <c1>1.0</c1>
  </integrator>

  <actuator name="asdf/act_clip">
   <output>asdf/act_clip_</output>
   <input>asdf/inp</input>
   <rate_limit>1.0</rate_limit>
   <clipto>
    <min>-1.0</min>
    <max>1.0</max>
   </clipto>
  </actuator>

  <actuator name="asdf/act">
   <output>asdf/act_</output>
   <input>asdf/inp</input>
   <rate_limit>1.0</rate_limit>
  </actuator>

 </channel>
~~~

1. Set `asdf/inp` to 10 and wait for some time, the unclipped integrator and actuator will go beyond the `<clipto>` values.
2. Set asdf/inp to -10.

Expectation: clipped integrator and actuator would immediately start going back from their limit.

Reality: they stay in sync with the unclipped versions, indicating that clipto is not respected by internal logic and just clips the output.

Octal450, 2019-08-14:
I have found this also. Same with <lag_filter> The value stops at the clip, but the calculation does not.

bcoconni, 2019-08-21:
> only the visible result is clipped

Does not seem like an issue to me but I'll check that in parallel to #212.

Octal450, 2019-08-22:
No this isn't what happens - what happens is that the calculation does not account for it. So for example in a lag_filter it will just hard stop at the clip value instead of clipping the input at it and then lagging.

Kind Regards,
Josh

mike402, 2019-08-22:
Maybe it's _intended_ to do that, just lacks documentation? But, personally, this would be counter-intuitive even if documented.

bcoconni, 2019-08-22:
> Maybe it's *intended* to do that, just lacks documentation?

Yes, `<clip>` clips the output and just does that. It is the last stage before copying the result to the output. See the last lines of code below.
https://github.com/JSBSim-Team/jsbsim/blob/851f2baf544a2a03c8ebe59a9731580e3c451817/src/models/flight_control/FGFilter.cpp#L144-L186
No wonder it is not behaving as you implied.
> But, personally, this would be counter-intuitive even if documented.

Intuition can be deceived since obviously `<clip>` does not do what you think it should. It does not mean however that the current code is wrong. Actually it has been so for the last 14 years at least (see the code below from commit bd905aa dating from Jun13, 2005):
https://github.com/JSBSim-Team/jsbsim/blob/bd905aa07d789c1716bde7cef5255c801a48e47b/src/models/flight_control/FGFilter.cpp#L125-L177
As you can check by yourself the sequence of instructions before returning has remained the same for 14 years:
* Clip
* Copy the result to the `<output>` properties
* Return

bcoconni, 2019-08-22:
> So for example in a lag_filter it will just hard stop at the clip value instead of clipping the input at it and then lagging.

This result can be obtained by the sequence of a `<pure_gain>` component followed by a `<lag_filter>`
```xml
<pure_gain>
  <input> test/X </input>
  <gain> 1.0 </gain>
  <clipto>
    <min> test/min </min>
    <max> test/max </max>
  </clipto>
  <output> test/Y </output>
</pure_gain>

<lag_filter>
  <input> test/Y </input>
  <c1> test/lag </c1>
  <output> test/result </output>
</lag_filter>
```

bcoconni, 2020-12-11:
This is not a bug in JSBSim but rather a misinterpretation of how `<clipto>` works.
Issue closed.

closed 2020-12-11 by bcoconni