JSBSim-Team/jsbsim #207: How to modify output data?

hovick, 2019-08-13:


hovick, 2019-08-13:
Ok, I have tried again by completely eliminating the code lines wich I dont want such as
`<velocities> ON </velocities>` and it worked
Before I simply changed these functions to description text by adding  however it wouldnt affect the output..not sure why, any ideas?
I now want to change the columns of each property that appear, lets say for the massprops property, I'd like to remove I_{xx} & I_{xy} columns but maintain the I_{xz}

seanmcleod, 2019-08-13:
You can use a combination of individual properties e.g.

```      
<property> accelerations/udot-ft_sec2 </property>
<property> accelerations/vdot-ft_sec2 </property>
```
and/or 'macros' like `<rates>`, `<forces>` etc. which output a pre-set set of properties.

```
<rates> ON </rates>
<forces> ON </forces>
```

bcoconni, 2019-11-16:
@hovick Do you consider your question answered ? If so, can we close this issue ?

bcoconni, 2019-11-30:
Issuer is not responding. Issue closed.

closed 2019-11-30 by bcoconni