JSBSim-Team/jsbsim #208: How to fly on a negative altitudes

alexander-kuptsov, 2019-08-13:
Dear all,

I would be much appreciative if you could help in following: if we launch an aircraft from a hill then flight relative altitude can be negative and it is normal, so the question is how to simulate such with JSBSIM?

Thank you in advance!

seanmcleod, 2019-08-13:
Which piece of software needs to handle negative altitudes relative to the take-off point? JSBSim or Ardupilot?

alexander-kuptsov, 2019-08-13:
> Which piece of software needs to handle negative altitudes relative to the take-off point? JSBSim or Ardupilot?

To my mind - both
For example, wrong relative altitude selected by pilot/operator may cause to crash while flying in mountains on a low altitudes - which is also the case for simulation

at the moment, i am not able to crash the aircraft - it is just landing, then I can tackoff again

seanmcleod, 2019-08-13:
Okay so your issue is you're not seeing any crash detection? 

It wasn't clear from your initial description what the issue with negative altitude is. For example if the aircraft takes off from FAOR which is at 5,000ft above sea level and flies down to FACT at sea level part of the flight on approach and landing to FACT is below the take-off altitude, i.e. negative relative to the take-off altitude. But what special handling do you expect from JSBSim for this example, assuming a normal landing, i.e. not a crash at FACT?

In terms of crash detection in JSBSim I only see the following in the landing gear code. Currently a message is simply printed but the simulation doesn't stop given that `fdmex->SuspendIntegration()` is commented out.

```c++
// Crash detection logic (really out-of-bounds detection)

void FGLGear::CrashDetect(void)
{
  if ( (compressLength > 500.0 ||
      vFn.Magnitude() > 100000000.0 ||
      GetMoments().Magnitude() > 5000000000.0 ||
      SinkRate > 1.4666*30 ) && !fdmex->IntegrationSuspended())
  {
    ostringstream buf;
    buf << "*CRASH DETECTED* " << fdmex->GetSimTime() << " seconds: " << name;
    PutMessage(buf.str());
    // fdmex->SuspendIntegration();
  }
}
```

alexander-kuptsov, 2019-08-13:
Actualy I have two seprate issues:
1. Flying on a negative relative altitudes. 
2. Crashing

The first is a major one) You absoluty right in your example with takeoff  at 5,000ft above sea and landing at let say 1.000ft ASL. So I cannot setup such. When I am using reset.xml as follows (169.26m ASL), the aircraft cannot fly below 169.26m because this is 0 relative altitude. I am assuming the simulated terrain is flat, and the aircraft just land on it when reach 0 relative altitude. 
my reset.xml:
<?xml version="1.0"?>
<initialize name="Start up location">
  <latitude unit="DEG"> %lat% </latitude>
  <longitude unit="DEG"> %lon% </longitude>
  <elevation unit="M"> 169.260000 </elevation>
  <altitudeAGL unit="M"> 0.3 </altitudeAGL>
  <vt unit="FT/SEC"> 0.0 </vt>
  <gamma unit="DEG"> 0.0 </gamma>
  <phi unit="DEG"> 0.0 </phi>
  <theta unit="DEG"> 12.959979 </theta>
  <psi unit="DEG"> 6.299994 </psi>
</initialize>

seanmcleod, 2019-08-13:
You can definitely have JSBSim start the aircraft at some altitude and have the aircraft fly below this starting altitude. For example I have a custom flight simulator instructor station setup where I can set the aircraft up at say 1,500ft ASL trimmed for approach and have the aircraft land at an airfield whose runway is 150ft ASL.

In my setup I have a custom `GroundCallback` implementation which returns the ground elevation for a given lat,lon via data retrieved from Prepar3D which has a DEM.

In your case you're setting the terrain elevation to a fixed value of 169m and you start at an AGL of 0.3m so yes you can't fly below the ground.

alexander-kuptsov, 2019-08-13:
is it possible to send ground elevation via properties set over network? or only possible solution is to user custom Callback implemetation?

seanmcleod, 2019-08-14:
Yep, take a look at the following property.

```c++
  PropertyManager->Tie("position/terrain-elevation-asl-ft", this,
                          &FGPropagate::GetTerrainElevation,
                          &FGPropagate::SetTerrainElevation, false);
```

And the implementation of `SetTerrainElevation`.

```c++
void FGPropagate::SetTerrainElevation(double terrainElev)
{
  double radius = terrainElev + VState.vLocation.GetSeaLevelRadius();
  FDMExec->GetGroundCallback()->SetTerrainGeoCentRadius(radius);
}
```

alexander-kuptsov, 2019-08-14:
Yea, thanks a lot!

I did a try a couple of days ago but with no luck - JSBSim just fell down. 
Could you please share more or less correct "reset.xml" to use property "position/terrain-elevation-asl-ft" during simulation process? in my case it didn't work out of the box
Maybe I send the value too frequent, or maybe other conditions or constraints should be taken into account

Anyway thank you for your answers!

seanmcleod, 2019-08-14:
The `reset.xml` sets the initial conditions. You would need to update `position/terrain-elevation-asl-ft` whenever the terrain elevation changes for the aircraft's current lat, lon.

bcoconni, 2019-08-15:
> In terms of crash detection in JSBSim I only see the following in the landing gear code. Currently a message is simply printed but the simulation doesn't stop given that fdmex->SuspendIntegration() is commented out.

https://github.com/JSBSim-Team/jsbsim/blob/4c0c0930b3871aeebbc29a24063d85155a7c83d8/src/models/FGLGear.cpp#L543-L558
Just for the record, this code has been implemented because the ground reaction forces can blow out if the aircraft hits the ground violently resulting in the aircraft bouncing in a very unrealistic way (sometimes it could be sent into orbit :smile:). This is by no means meant to detect the aircraft structure failure that most people imply when they use the word "crash".

The detection thresholds have been set to very high values because it was meant not to interfere with the aircraft model own crash detection functions. It was rather a kind of "last chance" detection if things were going really ugly and the aircraft model had not caught the problem. Indeed since commit a1360bab, the freeze has been turned off by Jon.

bcoconni, 2019-11-30:
@alexander-kuptsov
I infer that the issue is closed from the thumb up to the response from @seanmcleod.

closed 2019-11-30 by bcoconni