JSBSim-Team/jsbsim #209: Why max steer angle is set to 5° ?

nico1as, 2019-08-14:
Hi, 

I'm wondered why in aircraft.xml config file, the max steering angle is set to only 5°. Therefore, it is almost impossible to follow taxiways and runaways on an airport as the aircraft is not able to turn properly.

Thanks very much!

Nicolas

bcoconni, 2019-08-15:
> I'm wondered why in aircraft.xml config file, the max steering angle is set to only 5°.

Which aircraft are you referring to ? Where did you get it from ? Without further details it is not possible to answer your question.

nico1as, 2019-08-15:
Hi, 

For the a320 aircraft for exemple: 
https://github.com/JSBSim-Team/jsbsim/blob/4c0c0930b3871aeebbc29a24063d85155a7c83d8/aircraft/A320/A320.xml#L78

The max steer is set to 5°....may be there is a reason  ?

Tanks

bcoconni, 2019-08-15:
> The max steer is set to 5°....may be there is a reason ?

After using my favorite search engine, I found this document http://www.a320dp.com/A320_DP/landing-gear/sys-12.9.2.html, it seems that the max steering angle for the A320 depends on the ground speed:
> When using the tillers, the nose wheel steering angle is ±75° until approximately 20 knots groundspeed. As speed increases, the angle decreases progressively to 0° at 70 knots. When using the rudder pedals, the nose wheel steering angle is ±6° until approximately 40 knots groundspeed. As speed increases, the angle decreases progressively to 0° at 130 knots.

So it seems that the A320 model has been set to the max steering angle that is imposed when operating the nose gear with rudder pedals with a ground speed lower than 40 kts. You might want to follow the real aircraft behavior in which case you will need to implement a more elaborate system in the model.

Alternatively the value set in `A320.xml` might be the result of blindly copy-pasting this line of code from another aircraft as is very often done with JSBSim and FlightGear models.

nico1as, 2019-08-15:
Tanks a lot.
Have a good day.

closed 2019-08-15 by nico1as