JSBSim-Team/jsbsim #211: <clipto type="cyclic">: unexpected behaviour in negative domain

mike402, 2019-08-18:
Hi,

Sorry, I'd promised to test this sooner, but only now I've noticed something wrong with the new cyclic clipto.

~~~
 <property>test/in</property>
 <channel name="Test">
  <pure_gain name="test/out">
   <input>test/in</input>
   <gain>1.0</gain>
   <clipto type="cyclic">
    <min>0.0</min>
    <max>360.0</max>
   </clipto>
  </pure_gain>
 </channel>
~~~

1. in=50 out=50, ok
2. in=370 out=10, ok
3. in=460 out=100, ok
3. in=-10, expectation: out=350, reality: out=-10
4. in=-400, expectation: out=320, reality: out=-40
5. in=-460, expectation: out=260, reality: out=-100

So it looks like a modulo was used as mapping function instead of sawtooth:

~~~
Sawtooth:
1  /| /| /| /| /|
0 / |/ |/ |/ |/ |
   -2 -1 0 1 2
Modulo:
1    |\ |\  /| /|
0    | \| \/ |/ |
   -2 -1 0 1 2
~~~

As I said in some other issue, for correct mapping of negative inputs, the mapping function should absolutely be sawtooth instead of modulo -- that is, it should repeat the same way into negative domain -- but modulo is _mirrored_ in negative domain.

But it gets even weirder when you use a negative `<min>`, so the problem may be something else:

~~~
 <property>test/in</property>
 <channel name="Test">
  <pure_gain name="test/out">
   <input>test/in</input>
   <gain>1.0</gain>
   <clipto type="cyclic">
    <min>-180.0</min>
    <max>180.0</max>
   </clipto>
  </pure_gain>
 </channel>
~~~

1. in=50 out=50, ok
2. in=190 out=-170, ok
3. in=400 out=40, ok
4. in=-10 out=-10, ok
5. in=-190 expectation: out=170, reality: out=-190
6. in=-400, expectation: out=-40, reality: out=-400

bcoconni, 2019-08-21:
OK I'll check that. However I've already much on my plate so it might take some time before I can investigate this issue.

bcoconni, 2019-09-21:
I just pushed a fix to `master` (commit 0d61e02)

mike402, 2019-09-23:
Just tried the fix, works perfectly now -- thanks a lot!

closed 2019-09-23 by mike402