JSBSim-Team/jsbsim #212: pow(x, 1.5) gives NaN

NikolaiVChr, 2019-08-20:
Have I done something bad, or is this a bug? If I do power to 2, then there is no issues.

I cannot fly long time with this function before it starts sending NaN. The input property is between -1.0 and 1.0:

```xml
<fcs_function name="fcs/rudder-cmd-pow-norm">
            <function>
                    <ifthen>
                        <lt>
                            <p> fcs/rudder-cmd-norm </p>
                            <v> 0 </v>
                        </lt>
                        <product>
                            <v> -1 </v>
                            <pow>
                                <p>fcs/rudder-cmd-norm</p>
                                <v>1.5</v>
                            </pow>
                        </product>
                        <pow>
                            <p>fcs/rudder-cmd-norm</p>
                            <v>1.5</v>
                        </pow>
                    </ifthen>
            </function>
        </fcs_function>
```

bcoconni, 2019-08-21:
I suppose that this function crashes when `fcs/rudder-cmd-norm`is negative ? Indeed the square of a negative value is defined while the power of a negative value to 1.5 is mathematically undefined (and therefore generates a `NaN`).

If that is the case, I would suggest to apply the `<pow>` function to the absolute value of `fcs/rudder-cmd-norm`.

mike402, 2019-08-21:
Which reminds me, sometimes you have to use absolute value of some input with some functions and then use the sign of that input somewhere. Could a `<sign>` function be added to `<function>` in the future?

NikolaiVChr, 2019-08-23:
Yes, I was being stupid, sorry for wasting your time, will close the issue.

closed 2019-08-23 by NikolaiVChr