JSBSim-Team/jsbsim #213: static GroundCallback problem

SummersRain, 2019-08-20:
I created multiple FGFDMExec Pointers in the same process for multiple airplanes. Since GroundCallback in the FGLocation class is a static pointer, SetGroundCallback(0) is called when FGFDMExec objects are destructed, which causes other FGFDMExec objects to crash when destructed. On the other hand, when simulating multiple aircraft, a call to SetTerrainGeoCentRadius for the GroundCallback of one aircraft affects the others.

seanmcleod, 2019-08-20:
This is a known issue, take a look at - https://github.com/JSBSim-Team/jsbsim/issues/201

bcoconni, 2019-08-21:
> This is a known issue, take a look at - #201

Agreed, I suggest to continue the discussion in the issue #201 and this is issue is closed as a duplicate.

closed 2019-08-21 by bcoconni