JSBSim-Team/jsbsim #214: Atmospheric reentry

toopazo, 2019-08-22:
Hello guys, 

Is it possible to use JSBSim to simulate the Atmospheric reentry of a vehicle ??

I am looking for a tool to:
    - simulate 6DOF vehicles descending from 400km orbits down to earth's surface
    - incorporate thermal dynamics as well (ideally) 

Can I do this with JSBSim or somehow modify it to do so?

Thanks in advance

seanmcleod, 2019-08-22:
Take a look at this space shuttle project which makes use of JSBSim - http://wiki.flightgear.org/Space_Shuttle

bcoconni, 2019-08-22:
Just to complement @seanmcleod comment, the Space Shuttle FDM is one of the more complex (if not the most complex) JSBSim model out there. It is definitely a showcase of what JSBSim is capable of but you do not need to have such an elaborated model to simulate a vehicule reentry.

At the other end of the spectrum there is the canon ball which the simplest example : a ball falling through the atmosphere with a simple drag force
https://github.com/JSBSim-Team/jsbsim/blob/851f2baf544a2a03c8ebe59a9731580e3c451817/aircraft/ball/ball.xml#L1-L163
The complexity of your model will really depend on the level of fidelity you expect to get.

toopazo, 2019-08-22:
Thank you guys for your answers!

I read the  http://wiki.flightgear.org/Space_Shuttle link and it looks like the vehicle model (actuator modelling) itself is very involved and detailed ! cool!

But I guess my real doubt is how accurate is the atmospheric model of JSBSim at high altitudes (above 20km). For example MATLAB uses the International Standard Atmosphere model 
[T, a, P, rho] = atmosisa(height)
But it only covers from 0km and 20km

I was looking at https://jsbsim-team.github.io/jsbsim-reference-manual/mypages/formulation-manual-earth-modelling/ to find out what JSBSim but is not filled in yet

Thanks!

seanmcleod, 2019-08-23:
The atmosphere model is also based on the ISA model, but it does go up to 90km.

https://github.com/JSBSim-Team/jsbsim/blob/76411ec4d0a095ca031da320c466db3561c31267/src/models/atmosphere/FGStandardAtmosphere.cpp

dershow, 2019-08-23:
JSBSim also has NRLMSIS that can go much higher:

https://github.com/JSBSim-Team/jsbsim/blob/76411ec4d0a095ca031da320c466db3561c31267/src/models/atmosphere/FGMSIS.cpp <https://github.com/JSBSim-Team/jsbsim/blob/76411ec4d0a095ca031da320c466db3561c31267/src/models/atmosphere/FGMSIS.cpp>

https://ccmc.gsfc.nasa.gov/modelweb/models/nrlmsise00.php <https://ccmc.gsfc.nasa.gov/modelweb/models/nrlmsise00.php>

https://www.brodo.de/space/nrlmsise/index.html <https://www.brodo.de/space/nrlmsise/index.html>

I’m not sure if there is a flag to choose which one to use.

--Adam

toopazo, 2019-08-24:
Thank you

Indeed, something like NRLMSISE-00 is what I was looking for.

closed 2019-08-24 by toopazo

bcoconni, 2019-08-24:
> JSBSim also has NRLMSIS that can go much higher:

Theoretically the ISA model is defined up to 1000km, it is just that the equations above 91km are not yet coded in JSBSim.

> I’m not sure if there is a flag to choose which one to use.

At the moment the MSIS atmosphere is dead code in JSBSim so you cannot activate it.

Anyway the interface code is inaccurate because it assumes that the *"gas constant"* `R` is constant and equal to `1716.488` when deriving the sound speed and pressure from density and temperature:
https://github.com/JSBSim-Team/jsbsim/blob/851f2baf544a2a03c8ebe59a9731580e3c451817/src/models/atmosphere/FGMSIS.cpp#L169-L170
https://github.com/JSBSim-Team/jsbsim/blob/851f2baf544a2a03c8ebe59a9731580e3c451817/src/models/atmosphere/FGMSIS.cpp#L180
Unfortunately this assumption breaks when the molecular weight fractions vary with the altitude as is the case with the NRLMSIS-00 model.

Finally, maintenance-wise, the original C code has been manually inserted in a C++ class which makes it uneasy to keep synchronized with the official code. If the MSIS code was to be reactivated in JSBSim I'd rather use the official C source files verbatim and create a separate C++ class that would call that C code. That would make maintenance a breeze since updates of the MSIS code could be obtained by overwriting the C source files with their last update.