JSBSim-Team/jsbsim #217: Always a negative pitch moment disturb my aircraft

Octal450, 2019-09-16:
Hi,
I've had a bit of a bizarre situation here, I asked someone who is better with Aero than I and also he does not know - so let me explain:

I noticed that my PA28 is unable to rotate properly. I check CG location with my schematics from the real manuals, and its correct.

When I am at 65kts or 60kts in the air, the amount of elevator used is correct, and produces a positive pitch moment of a few degrees a sec as it does in the real plane. But when on ground, the aircraft does not rotate unless I pull back much much more. So, for test I moved CG aft even to a ridiculous amount - and still this problem. I must pull back much more almost 3 times as much to get the nose wheel off the ground! So I decided to cross out my aero as a factor so: 

1) I comment out every pitch moment in my aerodynamic except Cmde and Cmq (pitch elev, and pitch due to q)
2) I put every weight, fuel tank, and AeroRP at CG

But still same problem! Cannot rotate - even though nothing is stopping me now. When in air, correct pitching moment at the rotation elevator, but on ground it still get stuck right before nose wheel lifts off and I have to pull back harder. My theory is on this mysterious negative pitch moment that for some reason exists even when I have nothing else enabled and everything is at CG!

Check it out:
[image]

Wondering your thoughts.

Kind Regards,
Josh

andgi, 2019-09-16:
Have you double or triple checked that the landing gears are in the correct positions?

Octal450, 2019-09-16:
Yes I have. I am using VRP of 0,0,0 and I took the coordinates directly from the 3D model. I've had this issue in multiple planes, by the way.

seanmcleod, 2020-06-22:
@Octal450 did you manage to resolve this issue?

Octal450, 2020-06-22:
Nope. Still happens, I just ended up having to offset it with Cm0.

Kind Regards,
Josh

seanmcleod, 2020-06-22:
Can you share your PA28 JSBSim FDM?

Octal450, 2020-06-22:
https://sourceforge.net/p/flightgear/fgaddon/HEAD/tree/trunk/Aircraft/PA28/
https://sourceforge.net/p/flightgear/fgaddon/HEAD/tree/trunk/Aircraft/PA28/PA28-161-160.xml
https://sourceforge.net/p/flightgear/fgaddon/HEAD/tree/trunk/Aircraft/PA28/Systems/pa28-aerodynamics.xml

Kind Regards,
Josh

closed 2020-12-11 by bcoconni