JSBSim-Team/jsbsim #218: Initialization of running engines problem

bgaudron, 2019-09-27:
Hi,

I'm currently using the python wrapper to run simulations from a given state with a fixed initial throttle and mixture. 

So I call the `FGPropulsion::InitRunning` function just after setting and running ICs (including my initial throttles and mixtures ) but I see that in this function the throttles and mixtures are then overrided to 1:
https://github.com/JSBSim-Team/jsbsim/blob/0d61e0227882286dcf40afd9d8d98beb46362275/src/models/FGPropulsion.cpp#L346-L347

The engines are then initialized and running in a wrong state compared to the one set in ICs and it takes some time to reach the right given state.
So, would it be possible to remove those lines so that the user could be able to set itself the initial state of the engines ?

Thank you !

bcoconni, 2019-10-10:
Given that engines throttle and mixture positions are not part of the initial conditions, I need some clarifications about your request. Would you mind providing a small extract of your code so that I understand the sequence in which you are defining the initial conditions and then initializing the simulation ?

bgaudron, 2019-10-11:
An example would be : 
```
fdm = jsbsim.FGFDMExec(root_dir)

# Load aircraft
fdm.load_model('A320')

# A set of initial conditions
init_conditions = { 'ic/h-sl-ft': 10000,              # Initial position of the aircraft
                    'ic/long-gc-deg': 1.442031,
                    'ic/lat-geod-deg': 43.607181,

                    ...

                    'fcs/throttle-cmd-norm': 0.5,     # Initial state of the engines
                    'fcs/throttle-cmd-norm[1]': 0.5,
                    'fcs/mixture-cmd-norm': 1,
                    'fcs/mixture-cmd-norm': 1
}

# Set the initial conditions
for prop,value in init_conditions.items():
    fdm.set_property_value(prop,value)
fdm.run_ic()

# Set all engines running
fdm.propulsion_init_running(-1)
```

When I print the thrust of the engine with the current `FGPropulsion::InitRunning` and with the modified version where I remove the given lines, I get : 
[image]
So we can see that currently the engines are not initialized in the steady state given in the initial conditions and take some time to reach it because of the overrided throttles and mixtures.

bcoconni, 2019-10-12:
OK. Thanks for the example.

Unfortunately removing the lines you suggested would be a backward compatibility change : the method `FGPropulsion::InitRunning` is meant to start the engines and set them at full power. This can also be triggered by the property `propulsion/set-running` at any time.

Actually given that `FGPropulsion::InitRunning` does not do much (it sets the throttle and mixture position then call `FGEngine::InitRunning` and `FGPropulsion::GetSteadyState`), I'd rather expose these last two methods in the Python module to allow customized engine start up sequences.

https://github.com/JSBSim-Team/jsbsim/blob/ee5eb834b57d79a4f9d92bca446f1c4f7acb85d0/src/models/FGPropulsion.cpp#L343-L352

bgaudron, 2019-10-12:
Ok yes that would be great if we could have these two methods in the Python module.
Thanks !

bcoconni, 2019-10-12:
The required methods are now available in the Python module. You can download the required Python wheel package from the [Rolling Release v2019](https://github.com/JSBSim-Team/jsbsim/releases/tag/Rolling-release-v2019).

The engine startup sequence should now look like
```python
  propulsion = fdm.get_propulsion()

  for i in range(propulsion.get_num_engines()):
    fdm['fcs/throttle-cmd-norm[{}]'.format(i)] = 0.5
    fdm['fcs/mixture-cmd-norm[{}]'.format(i)] = 1.0
    propulsion.get_engine(i).init_running()

  propulsion.get_steady_state()
```
Let me know if that works for you.

bgaudron, 2019-10-14:
It works perfectly. Thank you !

closed 2019-10-14 by bgaudron