JSBSim-Team/jsbsim #219: function: <random> and <urandom>: fixed value throughout simulation

mike402, 2019-09-30:
A fellow aircraft dev has just found out `<random/>` and `<urandom/>` from `<function>` have the same value throughout the whole session, so it is impossible to make such things as random failures in terms of MTBF without external hacks.

But it would have been far more useful to have those funcs pick a new number every iteration, in that case if one would want the old behaviour they could use a `<switch>` that would latch the values on command.

Or at least, if too many aircraft rely on them staying the same now (but I don't think so) it could be easier to make some property that would trigger their regeneration when set to 1 and keep the old way when set to 0?

abassign, 2019-09-30:
I confirm this exactly if I insert the method <random> in the function I do not get the variation of the value both with <random> and with <urandom>

    <function name="systems/dragchute/get-rnd">
        <random/>
    </function>

I can not understand as it seems to me a mistake too obvious for not being already reported.

legoboyvdlp, 2019-09-30:
No need to do anything; just reinitialize the random seed per documentation:
http://www.cplusplus.com/reference/cstdlib/rand/
e.g.
`srand (time(NULL));`

mike402, 2019-09-30:
Seed defines the _sequence_ of random numbers, but the problem is that those functions only take the first random number out of the sequence and that's it, it never ever changes again. But of course it would be nice to have different seed for each run.

jonsberndt, 2019-09-30:
You need to be careful when considering this behavior. There is a distinction between randomizing a value for each simulation frame, and for providing an initial random value for something such as a set of monte carlo simulation runs. I haven’t looked at this code in a while, but may get a chance this weekend. 

Jon

mike402, 2019-09-30:
On Mon, Sep 30, 2019 at 09:25:55AM -0700, Jon S. Berndt wrote:
> You need to be careful when considering this behavior. There is a distinction between randomizing a value for each simulation frame, and for providing an initial random value for something such as a [...]

Of course, that's why it would probably need to be opt-in until maybe
it's clear none of the aircraft rely on old behaviour.

You would be able to freeze the ever-changing value easily with e.g.
`<switch>` if you need randomized constant instead of a sequence. But
the current behaviour does not let you get a sequence at all.

jonsberndt, 2019-09-30:
There are some simulation properties that can be set to control the sim. One of them is "simulation/randomseed". If I recall correctly, that can be set on the command line to any integer number and it will change the random number seed and you will get a different random number sequence. So, maybe that fixes part of the problem. But not all of the problem.

seanmcleod, 2019-09-30:
So here are the relevant lines from `FGFunction.cpp` for `random` and `urandom`. 

https://github.com/JSBSim-Team/jsbsim/blob/0d61e0227882286dcf40afd9d8d98beb46362275/src/math/FGFunction.cpp#L517-L526

So if the same number is being returned repeatedly during a JSBSim run then either both `GaussianRandomNumber()` and `rand()` are repeatedly returning the same number or `FGFunction` is caching a single call and return value.

jonsberndt, 2019-09-30:
> So if the same number is being returned repeatedly during a JSBSim run then either both GaussianRandomNumber() and rand() are repeatedly returning the same number or FGFunction is caching a single call and return value.

I believe this was all working at one point. And I also looked at the code. It *looks* right at first glance.

Jon

abassign, 2019-10-01:
I did a test for FGFS 19.2 on Ubuntu Linux machine, I modified FGFunction.cpp code and recompiled only with local archives. The code is this:

    } else if (operation == "urandom") {
      auto f = [](const decltype(Parameters)& p)->double {
                 printf("#### CIAO MONDO");
                 srand((unsigned int)time(0));
                 return -1.0 + (((double)rand()/double(RAND_MAX))*2.0);
               };

But the result is always the same, the value does not change.

I verified with a printf test that the random or urandom function is managed as a constant, such as pi ... and therefore can only produce a constant value!

My question is:
Why did we only realize this now?
First the random and urandom function modified its running value or was it always followed only once and then by a constant result?

If before it gave a constant result and we need a random number generator it might be more convenient to build a new block that generates random numbers, perhaps with some auxiliary parameter, such as a gain, a seed for having a pseudo-random sequence, and a trigger to get the pseudo-random number only when requested.

If anyone is interested I can write the generator as I have already done for the linear actuator.

Let me know your opinion.

mike402, 2019-10-02:
If I remember my math 101, there is a way to get any other Gaussian distribution from `<random>` and any other uniform distribution from `<urandom>` by adding/multiplying, so no need to write an extra module?

abassign, 2019-10-03:
mike402, right question, the reason comes from the fact that del <random> and <urandom> must release only a constant value, they do not serve as generators of random numbers for many applications. According to <random> and <urandom> they do not allow the insertion of a "seed" to create a predetermined random sequence, these functions do not allow a trigger that generates the random number at a certain event, in many applications, for example the generation of random numbers to get a shake effect is followed by a lag filter and a gain and a cut filter. For this there are all the reasons for having a complete module that may even introduce other types of random distributions.

mike402, 2019-10-03:
Do you need a trigger for certain event -- you can always use a `<switch>` on that event?

What random distributions can not be gotten by taking some func from the already implemented ones?

So `<random>` are `<urandom>` are _intended_ to return a fixed number?

About the insertion of seed, yes, I think that woud be useful (but probably only if own in each place and also if it's randoming on an event, not each step?)

seanmcleod, 2019-10-03:
> So `random` are `urandom` are intended to return a fixed number?

That's the first question to be answered. Based on @jonsberndt's comments it doesn't sound like the intended behavior is to return a single value for every call.

abassign, 2019-10-03:
@mike402 The idea of the trigger is to be able to generate a pseudo random sequence when necessary because JSBSim continuously executes the code (in FGFS 120Hz) and therefore it is impossible to have a pseudorandom test sequence. Normally the generator is activated only when it is necessary to make sure that the sequence is always the one required. The problem is in the use of the seed, it seems to me that the random C ++ generator has a unique seed definition, while if you run on an instantiated class you get a specific seed. https://en.cppreference.com/w/cpp/numeric/random

jonsberndt, 2019-10-03:
The way this is supposed to work is that each time the random function is present in XML code (each time that the random function is called) a new random number is supposed to be generated. For dispersing initial values, this is handled properly because it is only called once anyhow. In code that is executed each from, if the <random> construct is causing the same random number to be present each cycle, that would be wrong. But, the code doesn't look wrong to me. So, this is confusing.

seanmcleod, 2019-10-03:
Yep, so we need to debug what is happening.

So I added a `<urandom/>` element to a `<product/>` element in the Lift axis.

```xml
    <axis name="LIFT">
      <function name="aero/force/La_wing">
        <description> Wing lift due to alpha, regular air stream. </description>
        <product>
          <urandom/>
          <property> aero/qbar-psf </property>
          <property> metrics/Sw-sqft </property>
          <property> aero/coefficient/CLa_wing </property>
        </product>
      </function>
```

And then set a breakpoint in my debugger for the call to `rand()`.

And look where I ended up when I went back through the stack trace on the first hit of my breakpoint.

https://github.com/JSBSim-Team/jsbsim/blob/0d61e0227882286dcf40afd9d8d98beb46362275/src/math/FGFunction.cpp#L753-L759

So basically it's decided that it's a constant parameter and so optimizes it by replacing it with a single value.

So need to figure out why it thinks this is a constant function.

seanmcleod, 2019-10-03:
So on initial inspection, I'm not really familiar with the `FGFuntion` code, it looks like the `IsConstant()` method will return true if all the parameters return with `IsConstant() == true`, but it will always return true if there are no parameters which is the case for `urandom` in my example.

https://github.com/JSBSim-Team/jsbsim/blob/0d61e0227882286dcf40afd9d8d98beb46362275/src/math/FGFunction.cpp#L803-L811

seanmcleod, 2019-10-03:
There is actually a message printed out if `debug_lvl > 0` and `FGFunction` determines that the function is constant, although erroneously in the case of the random functions. 

```sh
In file ./aircraft/ah1s/ah1s.xml: line 965
<urandom> is applied on constant parameters.
It will be replaced by its result (-0.9975)
```

seanmcleod, 2019-10-04:
Below is a list of all the functions in `FGFunction.cpp`. It looks like all of them except `random` and `urandom` do return constant values if all their parameters are constant?

The random functions are also the only functions that can take 0 arguments?

So we could pass an extra argument when creating the random functions to indicate that they're not constant or modify `FGFunction::IsConstant()` to return false if there are no parameters, i.e. assume functions with no parameters aren't constant.

```sh
- random
- urandom

- product
- sum
- average
- difference
- min
- max
- or
- quotient
- pow
- toradians
- todegrees
- log2
- ln
- log10
- sign
- exp
- abs
- sin
- cos
- tan
- asin
- acos
- atan
- floor
- ceil
- fmod
- atan2
- mod
- fraction
- integer
- lt
- le
- gt
- ge
- eq
- nq
- not
- ifthen
- switch
- interpolate1d
- rotation_alpha_local
- rotation_beta_local
- rotation_gamma_local
- rotation_bf_to_wf
- rotation_wf_to_bf
```

mike402, 2019-10-04:
Btw, slightly unrelated question -- does
`<quotient><p>blablabla</p><v>123</v></quotient>` optimize into a
product with a constant -- how much would the performance savings be?

mike402, 2019-10-07:
Also, what do you think, maybe it would be nice to simply have a subtag for random and urandom that would make them change on every iteration and make it off by default?

seanmcleod, 2019-10-07:
> I can not understand as it seems to me a mistake too obvious for not being already reported.

It looks like this constant optimization was added fairly recently, Dec 27 2008 in the following commit - https://github.com/JSBSim-Team/jsbsim/commit/730a38505233b1d2f51692f12c5f51be44973b43

So it hasn't behaved this way for years, i.e. there shouldn't be lots of aircraft models/scripts that rely on this bug. 

So I suggest for now we simply fix it without introducing additional variations etc. of `random` and `urandom`.

mike402, 2019-10-07:
But maybe then the other way? An optional subtag or in-tag option for making it optimize to a constant?

seanmcleod, 2019-10-07:
> making it optimize to a constant?

Although it's not an optimization, you're asking for new functionality, i.e. the ability to tag a function as `ExecuteOnce`.

mike402, 2019-10-07:
If it doesn't optimize anything then it would be ok as it is (but the breakage fixed ofc).

bcoconni, 2019-10-14:
FYI, the @seanmcleod 's patch has been pushed to `master`and FlightGear.
Please test and let us know if that fixes the issue.

mike402, 2019-10-14:
On Mon, Oct 14, 2019 at 10:57:14AM -0700, Bertrand Coconnier wrote:
> FYI, the @seanmcleod 's patch has been pushed to `master`and FlightGear.
> Please test and let us know if that fixes the issue.

It seems to work here, thank you!

closed 2019-10-15 by seanmcleod