JSBSim-Team/jsbsim #220: Engines fixed directories

Octal450, 2019-10-07:
Hi - I am wondering if it can be known - why does the engine MUST be at Aircraft_DIR/Engines/myengine.xml?

Why Cannot be in a place I define - like everything else?

I would ask if it can be modified so that I can define my own path to engine.

Kind Regards,
Josh
PS: I am very tired at the moment, please let me know if I need more clarify.

mike402, 2019-10-07:
This has been bothering me as well -- why not use common system of paths with `<system include="blabla">` and such? Then engines and thrusters would be in the JSBSim subdirectory of the aircraft, together with all other systems.

Octal450, 2019-10-07:
Well it's actually an even simpler fix. Just add support for custom path define, just like every other file, system, aerodynamic, flight_control, etc

Kind Regards,
Josh

bcoconni, 2019-10-10:
JSBSim provides the ability to modify the folder in which the engines are loaded with the C++ function `JSBSim::FGFDMExec::SetEnginePath`  which is documented in [Developers docs](https://jsbsim-team.github.io/jsbsim/classJSBSim_1_1FGFDMExec.html#a407040feba3cfb275c6cc19e65f4cee5).

Octal450, 2019-10-10:
Hi Bertrand,
Perhaps I am just not seeing it, but how do I use this from the XML side?
Thanks.

Kind Regards,
Josh

bcoconni, 2019-10-12:
Sorry, I was in a rush the other day and my answer lacked some details.

Actually there is no such option as `include=...` neither for `<system>` nor for any other element. The option you are referring to is `file=...` and it is currently used to specify the file name of (or the path to) the engines definition. If the attribute `file=...` is provided with a relative path then JSBSim will look for files located under `Engines` or more generally under the engines default path (which is set to `engine` by default). This is the way that is almost always used in FlightGear to specify the path to engines definition files.

However, if the attribute `file=...` is provided with an absolute path then JSBSim can theoretically read a file that is located anywhere in the file system. This feature is available in FlightGear for a long time. If it does not currently work then it is a bug.

The last option is to modify the engine default path but this must be done before the aircraft model is loaded. As I mentioned earlier, there is already an option for that which is the C++ method `FGFDMExec::SetEnginePath`. For that option to be made available, you'd better ask the FlightGear devs to expose an option in FlightGear that allows to alter the JSBSim default engine path.

mike402, 2019-10-12:
Yes of course I meant `<system file="">`. I'm already using `<engine file="">` and it does not pick up engines from anything but the `Engines/` directory.

bcoconni, 2019-10-12:
@mike402 
Then your options are:
1. Provide an absolute path to `<engine file=...>`
2. Ask the FlightGear devs to provide an option to alter the default path `Engines` via the C++ method `FGFDMExec::SetEnginePath`.

closed 2020-12-11 by bcoconni