JSBSim-Team/jsbsim #221: Several Output files

fvarra, 2019-10-07:
Hello everyone,

I am pretty new in JSBSim and I have some doubts. I was trying to run the C172x with the autopilot (c172_cruise_8K.xml script) and I obtain 3 different output files when running the code. Why this is happening? Which one is the 'good' one?

Thank you all in advance

fvarra, 2019-10-07:
Actually I am obtaining 4, but the last one doesn't have any data. The files I obtain are:

c172_cruise_8K.csv
c172_cruise_8K_0.csv
c172_cruise_8K_1.csv
c172_cruise_8K_2.csv

bcoconni, 2019-10-10:
The script `c172_cruise_8K.xml` runs the same simulation with 3 different offsets for the ambient temperature (0, -36, 36). For each offset the simulation resets to test the new condition and that triggers the creation of a new instance of the output file.

The offsets of the ambient temperature `atmosphere/delta-T` are defined in the script https://github.com/JSBSim-Team/jsbsim/blob/0d61e0227882286dcf40afd9d8d98beb46362275/scripts/c172_cruise_8K.xml#L17-L26
and the reset logic is defined in the event "Reset". More specifically the reset is triggered at line 112 https://github.com/JSBSim-Team/jsbsim/blob/0d61e0227882286dcf40afd9d8d98beb46362275/scripts/c172_cruise_8K.xml#L103-L117
Since the script terminates just after the fourth reset (i.e. when `simulation/run_id` is higher than 3) a fourth output file `c172_cruise_8K_2.csv` is created but never used.

The relationship between he output files and the value of `atmosphere\delta-T` is summed up in the following table

| Output file | `atmosphere\delta-T` setting |
|-|-|
| `c172_cruise_8K.xml` | 0 |
| `c172_cruise_8K_0.xml` | -36 |
| `c172_cruise_8K_1.xml` | 36 |

bcoconni, 2019-11-16:
@fvarra Do you consider your question answered ? If so, can we close this issue ?

fvarra, 2019-11-18:
> @fvarra Do you consider your question answered ? If so, can we close this issue ?

Hello, thank you very much for your help, you saved me a lot of time. I still have to learn a lot about JSBSim.

Yes, we can close it,

Regards,

closed 2019-11-18 by fvarra